FD-/RPiPlay #1: Mirror stop

joelihn, 2019-07-25:
After building this project, i mirror from my iphone (iOS 12.4)  desktop to raspberry pi 3b+, but the picture is stuck at the image below.

Is there something i am doing wrong,thanks.

[image]

FD-, 2019-07-25:
Does rebooting the iPad fix the problem?

joelihn, 2019-07-26:
No,but when i reboot my iphone, it can be normal for a little while, and then stuck again.

FD-, 2019-07-26:
Please use the debug log flag, reproduce the problem and paste the complete log here.

FD-, 2019-07-26:
I think I found the problem: The mirroring thread doesn't properly handle read timeouts, so that it tries to receive a header when it got a timeout while receiving the payload. If the size of the payload data already received exceeded 128, the recv length parameter would underflow and cause invalid memory access. Please try if replacing the mirror thread function with this works:

'''
static THREAD_RETVAL
raop_rtp_mirror_thread(void *arg)
{
    raop_rtp_mirror_t *raop_rtp_mirror = arg;
	assert(raop_rtp_mirror);
	
    int stream_fd = -1;
    unsigned char packet[128];
    memset(packet, 0 , 128);
	unsigned char* payload = NULL;
    unsigned int readstart = 0;

    #ifdef DUMP_H264
    // C decrypted
    FILE* file = fopen("/home/pi/Airplay.h264", "wb");
    // Encrypted source file
    FILE* file_source = fopen("/home/pi/Airplay.source", "wb");
    FILE* file_len = fopen("/home/pi/Airplay.len", "wb");
    #endif

    while (1) {
        fd_set rfds;
        struct timeval tv;
        int nfds, ret;
        MUTEX_LOCK(raop_rtp_mirror->run_mutex);
        if (!raop_rtp_mirror->running) {
            MUTEX_UNLOCK(raop_rtp_mirror->run_mutex);
            break;
        }
        MUTEX_UNLOCK(raop_rtp_mirror->run_mutex);

        /* Set timeout value to 5ms */
        tv.tv_sec = 0;
        tv.tv_usec = 5000;

        /* Get the correct nfds value and set rfds */
        FD_ZERO(&rfds);
        if (stream_fd == -1) {
            FD_SET(raop_rtp_mirror->mirror_data_sock, &rfds);
            nfds = raop_rtp_mirror->mirror_data_sock+1;
        } else {
            FD_SET(stream_fd, &rfds);
            nfds = stream_fd+1;
        }
        ret = select(nfds, &rfds, NULL, NULL, &tv);
        if (ret == 0) {
            /* Timeout happened */
            continue;
        } else if (ret == -1) {
            logger_log(raop_rtp_mirror->logger, LOGGER_ERR, "raop_rtp_mirror error in select");
            break;
        }

        if (stream_fd == -1 && FD_ISSET(raop_rtp_mirror->mirror_data_sock, &rfds)) {
            struct sockaddr_storage saddr;
            socklen_t saddrlen;

            logger_log(raop_rtp_mirror->logger, LOGGER_DEBUG, "raop_rtp_mirror accepting client");
            saddrlen = sizeof(saddr);
            stream_fd = accept(raop_rtp_mirror->mirror_data_sock, (struct sockaddr *)&saddr, &saddrlen);
            if (stream_fd == -1) {
                logger_log(raop_rtp_mirror->logger, LOGGER_ERR, "raop_rtp_mirror error in accept %d %s", errno, strerror(errno));
                break;
            }

            // We're calling recv for a certain amount of data, so we need a timeout
            struct timeval tv;
[... 173 lines cut ...]
        }
    }

    /* Close the stream file descriptor */
    if (stream_fd != -1) {
        closesocket(stream_fd);
    }

    #ifdef DUMP_H264
    fclose(file);
    fclose(file_source);
    fclose(file_len);
    #endif

    // Ensure running reflects the actual state
    MUTEX_LOCK(raop_rtp_mirror->run_mutex);
    raop_rtp_mirror->running = false;
    MUTEX_UNLOCK(raop_rtp_mirror->run_mutex);

    logger_log(raop_rtp_mirror->logger, LOGGER_DEBUG, "raop_rtp_mirror exiting TCP thread");

    return 0;
}

'''

Wonderwuzi2000, 2019-07-27:
I want to confirm that your patch works well.
I had the same problem as described and made the changes as stated.
A video i took on my iPad now works fine - picture and sound are ok.

But - whenever i try to play locally stored music with the music app via Airplay to my TV i hear no sound and see the following error messages displayed:

aacDecoder_DecodeFrame error : 0x400c
aacDecoder_DecodeFrame error : 0x4002

Any idea how to fix this?

FD-, 2019-07-27:
@Wonderwuzi2000 thanks for the confirmation! For the aac decoder problem, please open a separate issue with as much details as you can give (iOS version, complete debug log etc).

closed 2019-07-27 by FD-

joelihn, 2019-07-29:
Yes, it works. Thank you.

By the way, is there some project support Android devices?

FD-, 2019-07-30:
RPiPlay is specifically and exclusively for the Raspberry Pi. If you want a similar Android project, have a look at @dsafa22's AirplayServer project (which RPiPlay is derived from).

joelihn, 2019-07-31:
> RPiPlay is specifically and exclusively for the Raspberry Pi. If you want a similar Android project, have a look at @dsafa22's AirplayServer project (which RPiPlay is derived from).

Sorry for my poor english...

I want to know is there some project can be run on Raspberry Pi like your project, and support android phone mirror ? not run on Android device.

FD-, 2019-07-31:
I see. There are two possibilities:

1. Write an Android app that uses the MediaProjection API to capture the screen content and implement the client side of the AirPlay protocol to send it to RPiPlay. This would have the disadvantage that DRM-protected content and apps that declare their content sensitive are not available through the MediaProjection API. There are a few such apps on the Play Store, though I don't know whether they are compatible with RPiPlay (I tested one and it used the ancient pre-iOS 9 AirPlay mirroring protocol, which is not supported by RPiPlay)

2. Write a Chromecast receiver program for the Raspberry Pi. This is definitely not a trivial task, as it seems the protocol is even less documented than AirPlay, so you'd have to reverse-engineer it all on your own.

None of these is in the scope of the RPiPlay project.

svarvel, 2019-11-06:
Android mirroring (and control) on the pi works really well with this tool: https://github.com/Genymobile/scrcpy