FD-/RPiPlay #3: After stop mirroring it still show the last frame

henrikhierl, 2019-07-31:
After stoping mirroring, the last frame is still displayed, until a new connection is established or the the program is closed. I used the latest version.

Expectet behavior:
Don't show the last frame.

Devices:
iPhone(IOS 12.4 and 12.3)
raspberry pi 3b+

Log:
....
Handling request TEARDOWN with URL rtsp://192.168.0.67/4282327910622381556
httpd receiving on socket 15
Connection closed for socket 15
Destroying connection
httpd receiving on socket 18
Connection closed for socket 18
Destroying connection

FD-, 2019-07-31:
This is expected behavior. You could try if calling the background drawing method from the video renderer's flush method (and making sure it's called when a connection is closed) works as you expected. If you have a working patch, please submit a pull request.

d0uub, 2019-08-16:
Kobajashi fixed the problem.

https://github.com/Kobajashi/RPiPlay

FD-, 2019-08-16:
I have now implemented this, based on my suggestion above. Kobajashi's implementation was a bit inelegant.

closed 2019-08-16 by FD-