FD-/RPiPlay #5: Raspbian non-headless server?

radokristof, 2019-07-31:


radokristof, 2019-08-01:
Sorry for the duplicate, I just renamed the title... Further investigating, it doesnt makes a difference if the desktop (xserver) is running or not. For me the output wont show anywhere. Where should I see the mirrored output?

FD-, 2019-08-01:
I tested RPiPlay on Raspbian full and on lite (headless). Both worked fine for me. Without additional details, I have no idea why its's not working for you. You don't mention OS version, program log, ...

radokristof, 2019-08-01:
iOS is the newest stable (12.4) and also the Raspian is the newest version (I can give you further info later today). Enabling the debug logging I cant see any error or whatsoever, it accepts the connection and I can see the received packets on the debug console... however I cant see it anywhere graphically

aykonsvk, 2019-08-11:
I have also the same issue RPI 4, iOS 12.4 newest Raspbian. I got this response in console:
`./rpiplay -n RPI -a hdmi`
`Initialized server socket(s)`
`*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.`
`*** WARNING *** Please fix your application to use the native API of Avahi!`
`*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>`
`Accepted IPv4 client on socket 15`
`Local: 192.168.2.108`
`Remote: 192.168.2.100`
`Accepted IPv4 client on socket 17`
`Local: 192.168.2.108`
`Remote: 192.168.2.100`
`raop_rtp_mirror starting mirroring`
`raop_ntp receive timeout`
`raop_rtp starting audio`

But when I run it with debug I can see that is is receiving video:
`raop_rtp_mirror video ntp = 1565522316639287, now = 1565522316634159, latency = -5128`
`Got h264 data of 7205 bytes`
`Video delay is -4705`

FD-, 2019-08-11:
What video driver are you using? How much gpu_mem is set? How many screens are you using?

aykonsvk, 2019-08-11:
I have everyhing by default. gpu_mem is 76 and driver I'm not sure about it. Only one screen. 1360x763 (hdmi) and now I tried it through vnc virtual desktop, (headless) and it is really low resolution. both cases same result

FD-, 2019-08-11:
Could you try setting gpu_mem to 256? Also, probably the non-standard resolution causes issues.

aykonsvk, 2019-08-11:
tried, but outcome was same. Btw I was little bit confused, becasue in `README.md` is stated to run executable `airplay_server` but there is only `rpiplay`

FD-, 2019-08-11:
Thanks for reporting the README mistake, fixed now. Regarding the display issue, are you using a stock unmodified system install? Do you probably have an older Pi to put the sd card in there and see if RPiPlay works? Also, since Pi 4 has two HDMI ports, I wonder if changing the used port fixes the issue.

aykonsvk, 2019-08-11:
Sure no problem, glad to help. Maybe another point, when I was searching for airplay on RPI, I found many people have issue with building it. Steps are clear, but ‘cmake ..’ is so confusing (a lot of them thought that command is missing. Maybe adding explanatory command or ‘../‘ would help. Unluckily I don’t have older RPI. Yes system is completely stock almost fresh installation. I will try later different hdmi port. But great project!

radokristof, 2019-08-11:
I thought of helping improving the README, after I can successfully start using AirPlay. 
Your problem seems the same as mine, though I have RPi 3B which only has 1 display output and I’m using it on a regular 1920x1080 display. 

However I still don’t know what the problem could be. How should it work? If I start mirroring a window should pop up with the mirrored screen or it appears as a full screen window?

FD-, 2019-08-11:
> I thought of helping improving the README, after I can successfully start using AirPlay. Your problem seems the same as mine, though I have RPi 3B which only has 1 display output and I’m using it on [...]

The whole screen should turn black and display the mirrored screen once a connection is made. The program uses OpenMAX for drawing, so it draws directly to the frame buffer, without involvement of the windowing system.

Well, that pretty much voids the two theories I had of it being caused by either the display or the Pi 4 hardware.
Please install a fresh unmodified Raspbian install as downloaded from the RPi website and see if the issue reproduces there. Also, it would be great to hear whether the same issue occurs for someone on a Pi 3B+ or Zero, since these are the only Pis I have for testing.

aykonsvk, 2019-08-12:
okay, I did a test today, tried HDMI 1 and HDMI 2 with Full HD and HD but the outcome was same with a clean Raspbian.

FD-, 2019-08-12:
Well, I'm really puzzled what's causing the issue then - RPiPlay works fine on a fresh Buster install on my Pi 3B+ and Pi 0.

radokristof, 2019-08-12:
For me, strech is installed not buster. Maybe this can be a problem...

FD-, 2019-08-12:
How did you test on a fresh OS install if you're not running Buster ?!

radokristof, 2019-08-12:
I didn't said I have reinstalled freshly this RPi. However I will try it with Buster.

FD-, 2019-08-12:
@radokristof Well, in one of your previous comments, you wrote that 'Raspian is the newest version'.

Either test it on a ___fresh___ install or provide a ___complete___ list of modifications you made to a ___fresh___ install. Otherwise, obviously, I cannot guarantee RPiPlay to work.

radokristof, 2019-08-12:
Ok sorry, I forgot about the new Buster. I will try with that soon.

radokristof, 2019-08-15:
Little update: I have updated my Raspberry Pi 3B to the latest buster. Now there is no package which can be updated. Also pulled this repo and rebuilt everything, however still the same error. Not a clean install yet, I will need to find an empty SD card to try it like that.

radokristof, 2019-08-21:
Got it working on a RPi 3B+ with Debian Strech on it. Works like a charm.
Only two questions remaining:

- Can I somehow detect if a mirroring is started? I would like to trigger my TV to start when mirroring is started....
- I'm also running Kodi on this Raspberry. If Kodi is running, I don't know why, but RPiPlay is not in foreground, it looks like if the mirrored video is half transparent...

radokristof, 2019-08-24:
Basically this is a different RPi. I tried to get it working with a 3B, this is a 3B+, and a headless Debian Strech is installed on it... I don't know which made a difference.
Anyway I already created a system service, so it can run in background/start on startup. Where we could place this in this repo? Add it to the README? I will submit a PR with it if you figure this out...

FD-, 2019-08-24:
I see. Does the same sd card work in the 3B as well?

Regarding the system service: Well, there already is this wiki page for auto-starting RPiPlay: https://github.com/FD-/RPiPlay/wiki/Starting-RPiPlay-at-system-start. IMHO, the advantage of starting it in the bashrc is that it can be easily stopped without remote access, just by pressing Ctrl+C. Since RPiPlay draws on top of everything else, it's difficult to stop or restart RPiPlay if it's started as a system service.

radokristof, 2019-08-24:
For me these Raspberry Pis are hard to access, in general I use ssh to control them. However I didn't see this article, nice solution!

I will try the same SD in the 3B as well.

closed 2019-09-05 by radokristof

cloudykeyru, 2020-02-13:
I had the same problem on RPI 3B. Has anyone been able to solve this problem?

myleshk, 2020-03-29:
Any updates for Buster? Got
```
raop_ntp receive timeout
raop_ntp exiting thread
```

FD-, 2020-03-29:
It's working fine for me on Buster. What log do you see?

myleshk, 2020-03-29:
> It's working fine for me on Buster. What complete log do you see?

```
pi@raspberrypi:~ $ ./RPiPlay/build/rpiplay -n MylesPi3 -a hdmi -b auto
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.1.100
Remote: 192.168.1.6
Accepted IPv4 client on socket 17
Local: 192.168.1.100
Remote: 192.168.1.6
raop_ntp receive timeout
```

And hangs... before I kill it.

Tried the following and nothing works:
- From Mac QuickTime video file
- From Mac QuickTime audio file
- From iOS Youtube video
- From iOS 3rd party video app

Only iOS Photos app can cast photos/videos but cannot hear sound.

myleshk, 2020-03-29:
The following log is produced under debug mode when casting from Mac QuickTime video:
```
pi@raspberrypi:~ $ ./RPiPlay/build/rpiplay -n MylesPi3 -a hdmi -b auto -d
> stream info: channel = 2	sample_rate = 44100	frame_size = 480	aot = 39	bitrate = 0
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.1.100
Remote: 192.168.1.6
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = 1934330068
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.1.100/13225986330028556825
DACP-ID: A92273BA7AFCE3B7
Active-Remote: 2065787230
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 62157
raop_ntp parse remote ip = 192.168.1.6
raop_ntp starting time
raop_rtp parse remote ip = 192.168.1.6
raop_rtp_mirror parse remote ip = 192.168.1.6
raop_ntp send_len = 32
eport = 37017, tport = 38953
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1585154282470428
Accepted IPv4 client on socket 17
Local: 192.168.1.100
Remote: 192.168.1.6
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = 1934330068
httpd receiving on socket 15
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.1.100/13225986330028556825
httpd receiving on socket 15
conn_request
Handling request RECORD with URL rtsp://192.168.1.100/13225986330028556825
raop_handler_record
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
raop_ntp send_len = 32
raop_ntp receive timeout
raop_ntp exiting thread
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request

...(and loops forever)
```

ArTyGill, 2020-03-29:
The same problem does not work on iphone 5c. Although the iPhone 11 and iPhone XR are all fine. There is also a problem when disconnecting, it is not possible to reconnect without restarting rpiplay.

Vatomatic, 2020-04-29:
Hello friends.
I am a Pi greenhorn! Just installed the RPiPlay over Buster on a B+ following the instructions to the letter down to the “make” command at the end in the README file. I cannot find an executable rpiplay file anywhere, what am I missing?

Thanks in advanced for your help.

headkit, 2020-10-29:
For me, changing the HDMI port on my RP4 did the trick. I now just have to fight a screen resolution problem.

Vatomatic, 2020-10-29:
> For me, changing the HDMI port on my RP4 did the trick. I now just have to fight a screen resolution problem.

Thanks!

pallas, 2020-10-30:
@headkit is the resolution problem on the rpiplay side or the monitor/tv or the mirroring device? Did you find the place to change the resolution in rpiplay?

headkit, 2020-11-02:
@pallas thanx for asking. I solved the problem by restarting the system and TV.

JYKeith123, 2021-12-25:
Same problem. I tested latest raspbian desktop version and lite version on my rpi 3b. I made the sd card with official imager. both not showing black screen but server logs only. I can hear audio but not video or screen. I started airplay on controlcenter and inside of an app like youtube as well with ipad pro gen 1 OS 14.8

JYKeith123, 2021-12-26:
#296 solved my prob :-)