FD-/RPiPlay #6: MacOS Mojave Mirroring doesn’t work

kevin-hv, 2019-08-04:


FD-, 2019-08-04:
I'm able to reproduce the problem and have traced it back to the max_dec_frame_buffering injection in the video renderer. I played around with the max_dec_frame_buffering value a bit, and it seems reducing it to 4 works with iOS (portrait and landscape) and mirroring a 1080p external monitor.

So, please modify [line 301 of renderers/video_renderer_rpi.c](https://github.com/FD-/RPiPlay/blob/c14be0f6270fdc744527cc6fb727f3efaf920b6a/renderers/video_renderer_rpi.c#L301) to:
```
h->sps->vui.max_dec_frame_buffering = 8;
```
And let me know if that helps!

Technical background:
The max_dec_frame_buffering injection is used to reduce video pipeline latency on the Raspberry Pi GPU. The idea is mentioned in a thread on the RPi forums, but since we don't have any control over the stream creation, we have to inject it into the SPS NAL at the receiver end, which is hacky on its own (And I don't really know why only certain values work, while values slightly higher or lower don't).  From my testing, the h264 library RPiPlay uses isn't 100% reliable, so that adds another potential error source.

While implementing max_dec_frame_buffering injection, for a while the code for constructing modified_data had a bug that I only noticed when I tested iOS portrait before publishing my code. The currently used max_dec_frame_buffering value 8 was tested with macOS and iOS while that bug was still present.

FD-, 2019-08-04:
@kevin-hv Can you confirm the fix?

kevin-hv, 2019-08-04:
Just Building it, sry I wasn’t available.
Commenting asap.

kevin-hv, 2019-08-04:
Now the mirroring works, thank you for the quick fix.

closed 2019-08-04 by kevin-hv