FD-/RPiPlay #8: Could not init video renderer

eriksmets, 2019-08-04:
`Error: Could not init video renderer.`

On latest Raspbian Buster and Raspberry Pi 4. 
How should I try and debug this ?

FD-, 2019-08-04:
What graphics driver are you using? As the relevant code doesn’t have much logging, I suggest you add a few printf calls to the video_renderer_init_decoder function (or use a debugger like gdb) to figure out what specific part in the video renderer initialization fails.

eriksmets, 2019-08-04:
fixed by increasing gpu_mem.

closed 2019-08-04 by eriksmets

sieroaoj, 2020-01-01:
I think the error log error could be improved with a better description:
 
rpiplay.cpp: function: start_server: line 234 
```
    if ((video_renderer = video_renderer_init(render_logger, show_background, low_latency)) == NULL) {
        LOGE("Could not init video renderer\n -> try to increase raspi-config > Advanced > Memory Split ");
        return -1;
    }
```

With this description people dont need to search this issue to solve the issue (my case).