FD-/RPiPlay #9: airplay without connect to wifi network

d0uub, 2019-08-14:
is it possible? 
i don't know why apple TV can do that..
raspberry pi can be hotspot but it requires extra step connect to wifi for my colleagues.

FD-, 2019-08-14:
There isn't any information available online about the P2P protocol AirPlay uses, and I'm not willing to reverse-engineer it. It probably makes use of AWDL, which has been reverse-engineered [here](https://owlink.org/wiki/). However, it seems like the open-source AWDL implementation doesn't work on Pi 3B+ and Pi 4.

I'll leave this open in case anyone wants to discuss a possible implementation.

btmcd, 2019-12-05:
> is it possible?
> connect without at the same network wifi

Are you looking for an audio only solution?

FD-, 2019-12-05:
@btmcd A solution like this is also trivially possible with RPiPlay. In fact, I've been using it for months. Just install RPiPlay and set up the Pi as a [WiFi access point](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md). However, AppleTVs support a different technology based on Apple's own take on the Wifi Hotspot/Direct protocol called AWDL that simplifies things from a user perspective (You don't have to manually connect to a WiFi network at all). Nevertheless, since I consider setting up a hotspot a reasonable solution and implementing AWDL would be an enormous task, I'm closing this issue now.

closed 2019-12-05 by FD-