FD-/RPiPlay #10: Error with cmake

emiel1234, 2019-08-15:
I get this when using cmake.
Any help?
Thanks

`- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.1.0k") 
You have called ADD_LIBRARY for library ilclient without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Configuring done
CMake Error: Cannot determine link language for target "ilclient".
CMake Error: CMake can not determine linker language for target: ilclient
-- Generating done
`

FD-, 2019-08-15:
It seems the ilclient library is not where it's supposed to be. What OS are you using?

emiel1234, 2019-08-15:
It is osmc, so that could probably be it. Is there any way to correct that?

FD-, 2019-08-15:
RPiPlay's build system requires the VideoCore libraries that Raspbian provides at the paths

* /opt/vc/include/ 
* /opt/vc/include/interface/vcos/pthreads 
* /opt/vc/include/interface/vmcs_host/linux 
* /opt/vc/src/hello_pi/libs/ilclient

Building fails with the error above if the OS you use doesn't provide these libraries. You could try to copy them from a Raspbian install to the locations listed above. Alternatively, you could copy them into any other location and modify the includes in [renderers/CMakeLists.txt](https://github.com/FD-/RPiPlay/blob/master/renderers/CMakeLists.txt) accordingly.

closed 2019-08-15 by FD-