FD-/RPiPlay #13: Custom resolution for ultra wide monitor

R0hitYadav, 2019-08-25:
Thanks a lot @FD- for your efforts! I'm trying to use RPiPlay with an ultra wide monitor but I'm unable to set a desired resolution.

**Device**: RPi 4
**GPU memory**: 256MB
**Casting device**: 2019 MBP 13
**Monitor**: LG Ultrawide @2560x1080

The resolution remains locked at 1080p regardless of the RPi HDMI port. The only difference being HDMI port 1 introduces black bars on the side and HDMI port 2 stretches the stream to fill the whole screen.

Unfortunately updating the variables `plist_t displays_0_width_node` and `plist_t displays_0_width_pixels_node` to `2560` under the file `lib/raop_handlers.h` didn't help, please advice.

FD-, 2019-08-26:
What are you expecting to see?

R0hitYadav, 2019-08-28:
I was hoping to bump up the resolution from 1920x1080 to the monitor's native resolution 2560x1080.

Not entirely certain as to how AppleTV handles such resolutions but going through a few forums it does seem like AppleTV does have issues with ultra-wide displays.

Thanks again, I'll close the issue.

closed 2019-08-28 by R0hitYadav