FD-/RPiPlay #14: Can't reconnect

TommySchwartz, 2019-08-31:
I can only connect and mirror screen one time (works great!). 
Connecting the second time only shows black screen.

if I want to reconnect I have to restart RPiPlay to be able to mirror again.

Don't know where and how to debug, but please provide directions and I will try my best.

duartesaraiiva, 2019-08-31:
Having the same problem

FD-, 2019-09-01:
For a useful bug report, please include the following:

* Steps to reliably reproduce the problem
* The (debug) output of RPiPlay at the time of malfunction

Please test if undoing the latest commit (commit 3449d878) fixes the issue.

duartesaraiiva, 2019-09-01:
Start the rpiplay, then connect, disconnect and reconnect. THe 2nd time it will fail.
For me, if i use the "-b" arg fixes this error

Debug code without the "-b" flag:
`> stream info: channel = 2	sample_rate = 44100	frame_size = 480	aot = 39	bitrate = 0
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.1.144
Remote: 192.168.1.140
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = 1933547732
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.1.144/3899814181647565792
DACP-ID: 446194CE3E1F8F37
Active-Remote: 1363207522
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 52137
raop_ntp parse remote ip = 192.168.1.140
raop_ntp starting time
raop_rtp parse remote ip = 192.168.1.140
raop_rtp_mirror parse remote ip = 192.168.1.140
raop_ntp send_len = 32
eport = 42767, tport = 55225
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1567353558529908
Accepted IPv4 client on socket 17
Local: 192.168.1.144
Remote: 192.168.1.140
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = 1933547732
httpd receiving on socket 15
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.1.144/3899814181647565792
httpd receiving on socket 15
conn_request
Handling request RECORD with URL rtsp://192.168.1.144/3899814181647565792
raop_handler_record
httpd receiving on socket 15
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.1.144/3899814181647565792
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.1.144/3899814181647565792
[... 352 lines cut ...]
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.1.144/11966295245646678080
raop_rtp_mirror video ntp = 1567363593979361, now = 1567363593903401, latency = -75960
Got h264 data of 5561 bytes
Video delay is -74926
raop_rtp_mirror tcp socket closed
httpd receiving on socket 15
conn_request
Handling request TEARDOWN with URL rtsp://192.168.1.144/11966295245646678080
raop_rtp_mirror exiting TCP thread
httpd receiving on socket 15
conn_request
Handling request TEARDOWN with URL rtsp://192.168.1.144/11966295245646678080
httpd receiving on socket 18
Connection closed for socket 18
Destroying connection
httpd receiving on socket 15
Connection closed for socket 15
Destroying connection
raop_ntp stopping time thread
raop_ntp exiting thread
raop_ntp stopped time thread
Stopping...
`
Undoing to the commit [https://github.com/FD-/RPiPlay/commit/77ffeffa994bf356e320006dbfe7523cc81eb695](77ffeff) works but when I stop the projection the last frame keeps until reconnect again

FD-, 2019-09-03:
Please let me know if the latest commit fixes the issue.

TommySchwartz, 2019-09-03:
I get a compiler error when running make: "no member named 'hidden'".

`[ 98%] Building C object renderers/CMakeFiles/renderers.dir/video_renderer_rpi.c.o
/home/pi/RPiPlay/renderers/video_renderer_rpi.c: In function ‘video_renderer_init’:
/home/pi/RPiPlay/renderers/video_renderer_rpi.c:282:13: error: ‘video_renderer_’ {aka ‘struct video_renderer_s’} has no member named ‘hidden’
     renderer->hidden = false;
             ^~
make[2]: *** [renderers/CMakeFiles/renderers.dir/build.make:76: renderers/CMakeFiles/renderers.dir/video_renderer_rpi.c.o] Fel 1
make[1]: *** [CMakeFiles/Makefile2:465: renderers/CMakeFiles/renderers.dir/all] Fel 2
make: *** [Makefile:84: all] Fel 2`

TommySchwartz, 2019-09-03:
Added under struct video_renderers_s
`bool hidden;`
just to see if it compiled, and it does.

Now RPiPlay reconnects as expected!
Bonus is that I now can switch between duplicate/extended display settings without any trouble. 
Thank you!

FD-, 2019-09-03:
Oh, seems like I only committed some of the changes I intended to. Should be fixed now.

duartesaraiiva, 2019-09-03:
Tested and confirm it’s working

FD-, 2019-09-03:
I just committed a cleaner fix, so please test that as well. Also, probably an even cleaner solution might be possible, depending of the result of the [discussion in the Raspberry Pi forums](https://www.raspberrypi.org/forums/viewtopic.php?p=1530307#p1530307).

TommySchwartz, 2019-09-03:
Works!

FD-, 2019-09-03:
Thanks for the confirmation! According to 6by9 in the forum, the current implementation is the expected way to do things, so I'll close this issue now.

closed 2019-09-03 by FD-