FD-/RPiPlay #17: Can not make

Deymon112, 2019-09-14:
[image]
Invalid format libbrcmglesv2.so . Help please. Try it on raspberry (raspbian) and ubuntu 18.04 on my PC.

FD-, 2019-09-15:
Does /opt/vc/lib/libbrcmGLESv2.so exist? On Ubuntu, it likely doesn't. On Raspbian, it should.

Deymon112, 2019-09-15:
Yes. I download it. I thought a problem with rights, set 777. It did not help

FD-, 2019-09-15:
What does that mean, you downloaded it? Where from? What OS are you running?

Deymon112, 2019-09-15:
https://github.com/FD-/RPiPlay/issues/10#issuecomment-521695708
From this post.
Raspbian stretch, Ubuntu 18.04

FD-, 2019-09-15:
On Stretch, the files are already present, so you shouldn't see the issue there unless you modified the system. In the comment linked above, I was referring to the ilclient library sources, which are then freshly compiled during RPiPlay's build. libbrcmglesv2.so is a pre-built static library, so it's likely not portable.

What exactly does your error message say (translated to English)?

Deymon112, 2019-09-15:
/opt/vc/lib/libbrcmGLESc2.so: error adding symbols: wrong file format
collect2: error: Id returned 1 exit status
CMakeFiles/rpiplay.dir/build.make:110: recipe execution error for purposes "rpiplay"
Make [2]: *** [rpiplay] Error 1
CMakeFiles/Makefile2:75: recipe execution error for purposes "CMakeFiles/rpiplay.dir/all"
make[1]: *** [CMakeFiles/rpiplay.dir/all] Error2
Makefile:83: recipe execution error for purposes "all"
make: *** [all[ Error2

FD-, 2019-09-15:
What Raspberry Pi model are you using?

Deymon112, 2019-09-15:
Virtualbox with raspbian

Deymon112, 2019-09-15:
I want to test it on my pc then buy raspberry

FD-, 2019-09-15:
That is the problem then. You're likely running this on an x86 CPU (most laptops or desktop computers) and the Raspberry Pi is an ARM CPU, so you cannot just copy binaries compiled for the Raspberry Pi to your computer and expect them to work.
Even if VirtualBox would emulate the ARM instruction set, the Broadcom VideoCore GPU cannot be emulated, and RPiPlay makes use of it for rendering.

Deymon112, 2019-09-15:
Oh... Sorry. Thanks for help. 
What model do you recommend for stable work?

FD-, 2019-09-15:
I only tested on a Pi Zero and a Pi 3B+, and of these two, the 3B+ performs better.

closed 2019-09-15 by FD-