FD-/RPiPlay #18: RPI 4 Video is not started after connection

aykonsvk, 2019-09-16:
I can start connection on RPI4 (clean newest Raspbian) and I can connect using iPhone X (12.4.1) But video frame is not showed. 

./rpiplay -n RPI -a hdmi
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.2.108
Remote: 192.168.2.100
Accepted IPv4 client on socket 17
Local: 192.168.2.108
Remote: 192.168.2.100
raop_rtp_mirror starting mirroring
raop_ntp receive timeout
raop_rtp starting audio

But when I run it with debug I can see that is is receiving video:
raop_rtp_mirror video ntp = 1565522316639287, now = 1565522316634159, latency = -5128
Got h264 data of 7205 bytes
Video delay is -4705

FD-, 2019-09-16:
This seems similar to issuse #5. What specific version of Raspbian are you running? Did you modify it in any way?

aykonsvk, 2019-09-16:
Yes, I already wrote there. But he closed it without resolving it in any way. I have Newest Version 10 (buster) It is completely fresh installation.

aykonsvk, 2019-09-16:
Closing this now. They released new update apparently so I tried After 2 weeks update-distro and now it is working smoothly!

closed 2019-09-16 by aykonsvk