FD-/RPiPlay #19: ISO Image and Autoupdate

Secara, 2019-09-16:
First of all, I would like to say a very cool project.
And thank you for closing this open source market gap.

It would be very cool to be able to download an ISO image of the current version that is preconfigured and, like many other projects, just need to be loaded onto an SD card to boot the RPi, also a service that runs in the background and to the Checking version and executing Updates if needed would also be desirable.

Thanks for the good work and keep it up.

FD-, 2019-09-17:
1) Pre-built image
This would require a rebuild every time the base distribution or RPiPlay is updated. I'm not willing to manually build such an image every time and don't have the knowledge required to set up an automated build pipeline. However, if you're willing to contribute one of the two, you're welcome to do so!

Instructions for automatically starting RPiPlay on system boot are available in the project wiki.

2) Autoupdate
The only sensible way for doing that would be via apt, and I don't know how third-party programs can be included there.

BLucky-gh, 2019-09-27:
You might be able to solve 1 by setting up a webhook or smth to check for new releases and then have a bash or any other type of script to build it, as long as you do not want to directly modify the os for some reason (optimization?), but even then in many cases it can be automated. then just cp the needed folders in place or have a raspi build it on itself, then you can convert it to iso however you want.

FD-, 2019-10-29:
I'm closing this since it has very low priority.

closed 2019-10-29 by FD-