FD-/RPiPlay #21: Password in airplay

slucas97, 2019-09-22:
Hello!

Is this possible insert a password in rpiplay? Only who knows the password mirroring to rpiplay.

You shoud add a PayPal donate!!!

Thanks by your job!

FD-, 2019-09-25:
While the AirPlay Mirroring protocol has support for PINs (aka device verification) at least (not sure about passwords), I'm not going to add support for them, because there's a whole lot of crypto stuff involved for (IMHO) little gain in functionality. For anyone interested in implementing this, a reference implementation is available here: https://github.com/funtax/AirPlayAuth.

I don't have a PayPal donation address, but if you want to give sth back and you know some C, I'd highly appreciate code contributions (features or bug fixes)!

DrKittens, 2020-10-07:
>because there's a whole lot of crypto stuff involved for (IMHO) little gain in functionality

Before looking at spending time on this could we confirm (as im not very accross airplay streaming vs airplay mirroring) would this functionality enable mirroring of applications that require DRM, eg apple movies, or is this a seperate issue?