FD-/RPiPlay #22: Can't quit when set to autostart

ungeschneuer, 2019-09-25:
I followed the instructions on your wiki page (https://github.com/FD-/RPiPlay/wiki/Starting-RPiPlay-at-system-start) to autostart RPiPlay but now I can't quit it with Ctrl + C anymore. I guess because the terminal is not taking the keys in the background.

FD-, 2019-09-25:
It's working fine for me. The described solution doesn't start the program in the background, so I'm not sure I understand your last sentence.

fhopfensperger, 2019-09-25:
I solved the background program thing with a cronjob when the raspberry boots (sleeps for 20 secs to make sure everything is up and running before `rpiplay` starts).
`crontab -e`
add the follwing line
`@reboot sleep 20;/home/pi/RPiPlay/build/rpiplay -n airplay -a hdmi -b > /dev/null 2>&1`

FD-, 2019-09-25:
@fhopfensperger But with that solution, Ctrl+C doesn't work for stopping RPiPlay, right?

fhopfensperger, 2019-09-25:
yes you're right, I use that command to kill `RPiPlay`, because the program is running in the background.
`kill $(ps aux | grep [r]piplay | awk '{print $2}')`

FD-, 2019-09-25:
@fhopfensperger I see. What I'm using is `killall -9 rpiplay`. That's a bit shorter and also works in the case that rpiplay has got stuck somewhere.

fhopfensperger, 2019-09-25:
even better. btw. thank you for this great software.

cropd, 2019-09-26:
You could also create a systemd service and start/stop the service or enable it to start it on system boot.

FD-, 2019-09-26:
@cropd Yeah, but stopping it without remote access would be more difficult while RPiPlay overlays the terminal (than the way described in the wiki).

cropd, 2019-09-26:
@FD- I see... my Pi runs RPiPlay behind my beamer – so remote access is needed – and it starts when powering my beamer on... love it! thank you for this project!!

closed 2019-10-18 by FD-