FD-/RPiPlay #24: Generic linux support?

Saren-Arterius, 2019-09-28:
Basically I have a x86 linux server attached to my TV. It's obvious the linux server does not have required hardware/firmware/libraries. Any solution to that?

FD-, 2019-09-28:
You are welcome to contribute a generic linux pair of renderers. Have a look at the renderers folder. The audio and video renderer interfaces in the header files should be abstract enough, so you'll just have to add code that renders the provided h264 and aac frames using whatever generic API Linux has to offer for that. I partly developed RPiPlay on macOS (better IDE than on Pi directly), the dummy renderers are just so the project compiles there.

Saren-Arterius, 2019-09-28:
> You are welcome to contribute a generic linux pair of renderers. Have a look at the renderers folder. The audio and video renderer interfaces in the header files should be abstract enough, so you'll [...]

I see, this would be a fun weekend project.

FD-, 2019-09-28:
You could probably make use of FFMPEG/Libav for decoding and rendering. If done properly, the same renderer could probably run on Linux and on Pi, using whatever decoding hardware is available. I considered using FFMPEG/Libav before, but I don't know anything about its API and couldn't find much samples/documentation.

Saren-Arterius, 2019-09-28:
Well, I am trying just writing the h264 data to socket or pipe and call `mpv` or `mplayer` to deal with that. Sorry I am getting too lazy

klattimer, 2019-10-16:
Is anyone actively working on this?

bttd, 2019-10-17:
Hi!

I’m also intrested in this!

FD-, 2019-11-12:
I'm not working on this, since I don't need it, but if anyone is willing to contribute a PR, I'd gladly accept it!

jedahan, 2020-01-14:
Starting to do some research, I'd like to get this working on macOS and linux.

Currently looking at libs to make this easier:
* gstreamer
* libmpv
* libav

jedahan, 2020-01-14:
# libmpv
## examples
* [simple streaming file example, single c file](https://github.com/mpv-player/mpv-examples/blob/master/libmpv/streamcb/simple-streamcb.c)
* [wxwidgets opengl cpp example](https://github.com/mpv-player/mpv-examples/blob/master/libmpv/wxwidgets_opengl/main.h)
* [cocoa opengl single m file example](https://github.com/mpv-player/mpv-examples/blob/master/libmpv/cocoa-rendergl/cocoa-rendergl.m)

I will likely need guidance on how to incorporate the .m file into the cmake build system, I am very unfamiliar with it.

jedahan, 2020-01-14:
# libav / ffmpeg
## examples
* [video decoding (mpeg1) in c](https://github.com/FFmpeg/FFmpeg/blob/master/doc/examples/decode_video.c)

jedahan, 2020-01-14:
# gstreamer
## docs
* [gstvideosink](https://gstreamer.freedesktop.org/documentation/video/gstvideosink.html?gi-language=c)
* [application documentation](https://gstreamer.freedesktop.org/documentation/application-development/index.html?gi-language=c)

FD-, 2020-01-14:
From my research, it seems like ffmpeg does not include a video sink (renderer) component, so it can only be used for the decode and would require an rendering component, likely based on OpenGL and/or SDL.

It seems like libmpv is essentially ffmpeg + the rendering, though it appears more like a video player with an API for controlling it. I'm not sure how you can feed raw AAC and H264 data (plus timestamps!) into it and have it render the decoded frames. 

As for gstreamer, its pipeline architecture looks promising. This probably offers exactly the level of abstraction we need. Apparently, there's also an ffmpeg backend available for it.

From my perspective, I'd prefer a library that also has a backend for the Raspberry Pi's decoding hardware. The current OpenMAX based renderer AFAICT is as performant as it can get, but it's severely limited in the audio sinks it supports and is deprecated by the Pi Foundation.
I think both ffmpeg and gstreamer tick that box, though it seems like ffmpeg is preferred by the Pi Foundation.

While I don't have the time for pulling this off myself, I'd happily provide any information or assistance you need!

nook24, 2020-01-14:
After reading the docs it sounds like `gstreamer` should get the job done.
I tried to create a gstreamer based linux renderer for a couple of hours now. Without any luck so far..

I guess my C skills are to bad for this

jedahan, 2020-01-15:
I found this, seems interesting. https://github.com/dabrain34/gst-airplay

FD-, 2020-01-15:
AFAICT, this is a wrapper around shairport and thus only works for audio (if it does at all - shairport is no longer actively developed, only the shairport-sync fork is).

klattimer, 2020-01-28:
I took a shot at this a while back, I think it's not that hard to pull off... The PI just throws blocks of mpeg data at a renderer and hopes for the best it seems so gstreamer should be upto the job.

klattimer, 2020-02-02:
I just got round to doing a little testing, I managed to create a video renderer which dumps the buffer to disk, I then pointed gst-launch-1.0 at the file it dumped and it played perfectly fine... 

So all I need to do now I write up the appropriate gstreamer buffer decoder for video - obviously it'd be nice to support audio, but I'm content to live with just video myself right now...

klattimer, 2020-02-02:
This might be useful information

```
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: ring-buffer-max-size = 0
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: buffer-size = -1
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: buffer-duration = -1
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: use-buffering = false
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: download = false
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: uri = file:///home/kodi/Projects/RPiPlay/build/DUMPSTREAM.MP4
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: connection-speed = 0
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: source = "\(GstFileSrc\)\ source"
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = video/x-h264, stream-format=(string)byte-stream
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = NULL
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, width=(int)1804, height=(int)1016, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high, level=(string)4.1, codec_data=(buffer)01640029ffe1001e27640029ac1316501c4081ee5c01a808080d2606000bb80002ee0bdef82801000428ee1f2c
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/avdec_h264:avdec_h264-0.GstPad:sink: caps = video/x-h264, width=(int)1804, height=(int)1016, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high, level=(string)4.1, codec_data=(buffer)01640029ffe1001e27640029ac1316501c4081ee5c01a808080d2606000bb80002ee0bdef82801000428ee1f2c
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-h264, width=(int)1804, height=(int)1016, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high, level=(string)4.1, codec_data=(buffer)01640029ffe1001e27640029ac1316501c4081ee5c01a808080d2606000bb80002ee0bdef82801000428ee1f2c
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-h264, width=(int)1804, height=(int)1016, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high, level=(string)4.1, codec_data=(buffer)01640029ffe1001e27640029ac1316501c4081ee5c01a808080d2606000bb80002ee0bdef82801000428ee1f2c
/GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/avdec_h264:avdec_h264-0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1804, height=(int)1016, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt709, framerate=(fraction)0/1
/GstPlayBin:playbin0/GstInputSelector:inputselector0.GstSelectorPad:sink_0: always-ok = false
/GstPlayBin:playbin0/GstInputSelector:inputselector0.GstSelectorPad:sink_0: active = true
/GstPlayBin:playbin0/GstInputSelector:inputselector0: active-pad = "\(GstSelectorPad\)\ sink_0"
/GstPlayBin:playbin0/GstInputSelector:inputselector0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1804, height=(int)1016, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt709, framerate=(fraction)0/1
/GstPlayBin:playbin0/GstPlaySink:playsink.GstGhostPad:video_sink.GstProxyPad:proxypad3: caps = video/x-raw, format=(string)I420, width=(int)1804, height=(int)1016, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt709, framerate=(fraction)0/1
/GstPlayBin:playbin0/GstPlaySink:playsink/GstStreamSynchronizer:streamsynchronizer0.GstPad:src_0: caps = video/x-raw, format=(string)I420, width=(int)1804, height=(int)1016, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt709, framerate=(fraction)0/1
/GstPlayBin:playbin0/GstPlaySink:playsink/GstBin:vdbin.GstGhostPad:sink.GstProxyPad:proxypad7: caps = video/x-raw, format=(string)I420, width=(int)1804, height=(int)1016, interlace-mode=(string)progressive, multiview-mode=(string)mono, multivie [... cut ...] action)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt709, framerate=(fraction)0/1
/GstPlayBin:playbin0/GstInputSelector:inputselector0.GstSelectorPad:sink_0: tags = taglist, video-codec=(string)"H.264\ \(High\ Profile\)";
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "playbin0".
Execution ended after 0:00:08.996463236
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

klattimer, 2020-02-04:
!! PROGRESS !! 

It's not complete, in fact it puts out about 1 frame, ish... but it's definitely pumping the pipeline 

https://github.com/klattimer/RPiPlay/commit/ec94c6ee1384142d3692b948c0efebb3edb6bc8e

I think the issue somewhere in the timing and related stuff, duration and timestamp seem to be out of whack and I have not much of a clue how to get the audio muxed in, this will likely require some jiggery pokery with the pipeline.

klattimer, 2020-02-04:
I need someone with cmake skills to update the build stuff there so if GST is around use that, as gstreamer uses OpenMAX if available.

I think we could merge down all the video renderers into a gstreamer pipeline which has runtime optimisation and hardware selection. 

The following gst-launch command works for dumped video, I haven't captured dumped audio yet. 

gst-launch-1.0 filesrc location=./DUMPSTREAM.MP4 ! queue ! avdec_h264 ! videoconvert ! autovideosink

Saren-Arterius, 2020-02-04:
@klattimer I found that audio data will be written to the dump file only when there is a media playing, however it's raw AAC data that I have no idea to play it...

klattimer, 2020-02-04:
@Saren-Arterius try pumping it through a playbin pipeline, gstreamer might figure it out, or tell you which codec is missing at least.

FD-, 2020-02-04:
Thanks for your efforts! I'm enthusiastically following your progress! I'm afraid I don't know much about cmake either, but I remember it was pretty easy to pull off the current structure just following the example from their docs. 

Indeed, there only is audio data (and an audio connection, in fact) while the device is actually playing audio. The format is AAC-ELD.

Legodev, 2020-05-15:
I tried the source if @klattimer  (merged with current master + for testing purposes i also rename the output file to DUMPSTREAM.264)

This is not much Progress, but by creating a fifo `mkfifo DUMPSTREAM.264` i then via: `ffplay -framerate 120 -probesize 512 DUMPSTREAM.264` its almost usable. 

The high Framerate is to force ffplay to display the frames as they come in, -probesize reduces the delay until the first frame is shown. The latency is around 1 second.

Now the interesting part: when i add "-sync ext" then it hangs similar to the gstreamer output at the first frame. It then does continue but it hangs as well.

So my guess is the gstreamer Problem maybe be caused by a missing sync trigger. Only started looking into this a few hours ago, so did not have time read into the gstreamer api, but i still see a chance.

Waester, 2020-11-30:
Just want to say that @dougg3 PR is working. Maybe this can be closed now?

klattimer, 2020-11-30:
Happily :)

closed 2020-11-30 by FD-