FD-/RPiPlay #25: No video with iOS13 / iPadOs (pi0w)

aethernet, 2019-10-01:
Hi

First let me say that it's an amazing project!

I'm currently trying to cast video from an iPad Pro (v2) (iOS 12.3) but without success. 
Also tried my iPhone 7+ (iOS 13.1) and iPad 2018 (13.1) without success.

But I've been able to cast from my macbook air (Mojave 10.14.6) and it works great. So i guess my install of RPiPlay is alright.

Console doesn't give much info, but the Airplay menu on the device say : "iPad -> RPiPlay" Music and show the Apple Music icon even if i never use that app (I'm a Spotify user).

I tried with and without the hdmi audio output as my usecase would actually not require any sound just screen mirror.

Here's the log when running with debug (Spotify playing on my iPhone) :

Any idea ? 

```
pi@raspberrypi:~/RPiPlay/build $ ./rpiplay -b -a hdmi -d
> stream info: channel = 2	sample_rate = 44100	frame_size = 480	aot = 39	bitrate = 0
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.12.24
Remote: 192.168.12.36
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = -1286620972
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.12.24/11464370087663912438
DACP-ID: 2AA641442BB321A
Active-Remote: 2716497495
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 62322
raop_ntp parse remote ip = 192.168.12.36
raop_ntp starting time
raop_rtp parse remote ip = 192.168.12.36
raop_rtp_mirror parse remote ip = 192.168.12.36
eport = 33759, tport = 55594
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1569863837653087
Accepted IPv4 client on socket 17
Local: 192.168.12.24
Remote: 192.168.12.36
httpd receiving on socket 15
conn_request
[... 1034 lines cut ...]
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
httpd receiving on socket 15
conn_request
Handling request POST with URL /feedback
raop_handler_feedback
```

aethernet, 2019-10-01:
should have said, but the screen connected to the pi stays black

FD-, 2019-10-01:
Please confirm you are _not_ trying to use RPiPlay for casting audio only / casting a video. RPiPlay only implements the AirPlay mirroring protocol.

aethernet, 2019-10-01:
I'm trying to mirror my iPad Pro (iOS12.3) screen (i don't care about sound, just image is fine)

Everything else is test i did trying to understand what's wrong.

aethernet, 2019-10-01:
Works well when mirroring my laptop screen, stays black with iPad Pro (iOS 12.3) / iPhone 7+ (iOS 13.1) and iPad 2018 (iPadOS 13.1)

aethernet, 2019-10-01:
oh no!
I feel so stupid. I was using the audio output switch, as i'm used to airplay audio only :D

It works fine with the screen sharing button :)

Thanks so much for your amazing work (and incredibly quick support)

FD-, 2019-10-01:
Duplicate of #2

closed 2019-10-01 by FD-