FD-/RPiPlay #27: 3rd party Airplay streamers do not work

geekpersonman, 2019-10-04:
Have successfully compiled and installed RPiPlay on my raspberry pi 3b+ using a fresh install of buster off the raspberry pi website. It works great from my iPhone on iOS 13 and my MacBook Pro using Mojave, but the program fails when I try to stream using Airparrot 2 on my Windows 10 laptop. Here is the output log:
`Initialized server socket(s)`
`*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.`
`*** WARNING *** Please fix your application to use the native API of Avahi!`
`*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>`
`Accepted IPv4 client on socket 15`
`Local: 100.72.41.47`
`Remote: 100.72.41.41`
`raop_rtp_mirror starting mirroring`
`Accepted IPv4 client on socket 18`
`Local: 100.72.41.47`
`Remote: 100.72.41.41`
``rpiplay: /home/pi/RPiPlay/lib/raop_rtp_mirror.c:303: raop_rtp_mirror_thread: Assertion `nc_len > 0' failed.``
after that the program exits.

FD-, 2019-10-04:
Duplicate of #15

geekpersonman, 2019-10-04:
sorry I didn't see that issue before, Thanks! I was already working on getting Miracast going too, I was just hoping to run as little software on the pi as possible.

FD-, 2019-10-04:
Since 3rd party clients likely implement the best documented version of the protocol, it probably isn't too much effort to add it into RPiPlay. I won't do it because I don't need it, but if you are willing to invest some time, I'd be happy to help where I can.

geekpersonman, 2019-10-04:
Thank you, I'll look into it. I know Kodi has the old version of Airplay mirroring protocol builtin, so maybe most of that code could be reused? Also on an unrelated note I want to add libcec support so my tv will automatically turn on and change the input when a stream connects and turn off when it disconnects.

closed 2019-10-04 by geekpersonman

FD-, 2019-10-04:
If you have a working solution for any of these points, please submit a PR! Adding libcec support should likely be as easy as adding a few lines to the video renderer. flush is called when the stream ends.

geekpersonman, 2019-10-04:
What IDE are you using to write this? I am currently working in Xcode but I am struggling to get libcec to register in it properly.

FD-, 2019-10-05:
For most of the code I used a simple text editor, but I later switched to CLion (I have a student license), since it provides good support for CMake.

geekpersonman, 2019-10-05:
Thank you that’s very useful, I had looked into Clion because I love IntelliJ but I didn’t realize it was free for students.