FD-/RPiPlay #30: Error after make command

GhostR111der, 2019-10-10:
Hello!
I enter these commands
sudo apt-get install cmake
sudo apt-get install libavahi-compat-libdnssd-dev
sudo apt-get install libssl-dev
mkdir build
cd build
cmake ..
make

And after make command I see this:

renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/build.make:62: recipe for target 'renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/h264_stream.c.o' failed
make[2]: *** [renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/h264_stream.c.o] Error 1
CMakeFiles/Makefile2:569: recipe for target 'renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/all' failed
make[1]: *** [renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

And RPiPlay not starting! What I need to do to fix it?

FD-, 2019-10-10:
Are you sure this is the whole output? There is no indication on why it failed.

GhostR111der, 2019-10-10:
This is all output:

pi@raspberrypi:~/RPiPlay/build $ make
[  0%] Building C object renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/h264_stream.c.o
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘read_scaling_list’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:466:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int j = 0; j < sizeOfScalingList; j++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:466:5: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘read_svc_vui_parameters_extension’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:564:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int i = 0; i <= sps_svc_ext->vui.vui_ext_num_entries_minus1; i++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘read_hrd_parameters’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:676:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int SchedSelIdx = 0; SchedSelIdx <= hrd->cpb_cnt_minus1; SchedSelIdx++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘read_pic_parameter_set_rbsp’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:726:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for( int i_group = 0; i_group <= pps->num_slice_groups_minus1; i_group++ )
             ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:733:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for( int i_group = 0; i_group < pps->num_slice_groups_minus1; i_group++ )
             ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:749:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for( int i = 0; i <= pps->pic_size_in_map_units_minus1; i++ )
             ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:780:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for( int i = 0; i < 6 + 2* pps->transform_8x8_mode_flag; i++ )
             ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘write_scaling_list’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:1795:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int j = 0; j < sizeOfScalingList; j++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘write_svc_vui_parameters_extension’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:1893:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int i = 0; i <= sps_svc_ext->vui.vui_ext_num_entries_minus1; i++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘write_hrd_parameters’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:2005:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int SchedSelIdx = 0; SchedSelIdx <= hrd->cpb_cnt_minus1; SchedSelIdx++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘write_pic_parameter_set_rbsp’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:2055:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for( int i_group = 0; i_group <= pps->num_slice_groups_minus1; i_group++ )
             ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:2062:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for( int i_group = 0; i_group < pps->num_slice_groups_minus1; i_group++ )
             ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:2078:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for( int i = 0; i <= pps->pic_size_in_map_units_minus1; i++ )
             ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:2109:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
             for( int i = 0; i < 6 + 2* pps->transform_8x8_mode_flag; i++ )
             ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘read_debug_scaling_list’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:3124:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int j = 0; j < sizeOfScalingList; j++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘read_debug_svc_vui_parameters_extension’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:3222:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int i = 0; i <= sps_svc_ext->vui.vui_ext_num_entries_minus1; i++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘read_debug_hrd_parameters’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:3334:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for( int SchedSelIdx = 0; SchedSelIdx <= hrd->cpb_cnt_minus1; SchedSelIdx++ )
     ^
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function ‘read_debug_pic_parameter_set_rbsp’:
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:3384:13: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
[... cut ...]
renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/build.make:62: recipe for target 'renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/h264_stream.c.o' failed
make[2]: *** [renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/h264_stream.c.o] Error 1
CMakeFiles/Makefile2:569: recipe for target 'renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/all' failed
make[1]: *** [renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

FD-, 2019-10-10:
Did cmake succeed before that?

GhostR111der, 2019-10-10:
Yes

FD-, 2019-10-10:
Did you use git clone for getting the project files?

GhostR111der, 2019-10-10:
Yes, I use this commands
git clone https://github.com/FD-/RPiPlay.git
cd RPiPlay

FD-, 2019-10-10:
Please try if adding

    set (CMAKE_C_STANDARD 11)

after 

    set (CMAKE_CXX_STANDARD 11)

In the top-level CMakeList.txt fixes the problem.

FD-, 2019-10-10:
You must be using an old version of GCC. What system are you running and what gcc version do you use?

GhostR111der, 2019-10-10:
> Please try if adding
> 
> ```
> set (CMAKE_C_STANDARD 11)
> ```
> 
> after
> 
> ```
> set (CMAKE_CXX_STANDARD 11)
> ```
> 
> In the top-level CMakeList.txt fixes the problem.

Its not help. I check that I have GCC version 4.9.2

FD-, 2019-10-10:
That's the problem then. Why are you using this old gcc version? What system are you using?

GhostR111der, 2019-10-10:
I don't know) Im not a pro user. I have Raspbian GNU/Linux 8 (jessie)

FD-, 2019-10-10:
You don't need to be a pro user to keep your system updated. Jessie is almost 5 years old.

closed 2019-10-10 by FD-

GhostR111der, 2019-10-10:
How I can update my system? I make sudo apt-get update and upgrade options yesterday.

FD-, 2019-10-10:
Major OS updates require a reflash. Google is your friend.

GhostR111der, 2019-10-10:
Thank your very much!)

GhostR111der, 2019-10-10:
I upgrade my Raspberry to (stretch) and version of GCC now is 6.3.0. I tried to install RPiPlay again, and have new errors:

pi@raspberrypi:~ $ cd RPiPlay
pi@raspberrypi:~/RPiPlay $ cd build
pi@raspberrypi:~/RPiPlay/build $ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/RPiPlay/build
pi@raspberrypi:~/RPiPlay/build $ make
[  2%] Built target h264-bitstream
[  2%] Built target curve25519
[  7%] Built target ed25519
[ 10%] Built target playfair
[ 15%] Built target plist
[ 23%] Built target airplay
[ 24%] Built target ilclient
[ 97%] Built target fdk-aac
[ 99%] Built target renderers
[100%] Linking CXX executable rpiplay
lib/libairplay.a(crypto.c.o): In function `aes_reset':
crypto.c:(.text+0x35c): undefined reference to `EVP_CIPHER_CTX_reset'
lib/libairplay.a(crypto.c.o): In function `sha_init':
crypto.c:(.text+0x754): undefined reference to `EVP_MD_CTX_new'
lib/libairplay.a(crypto.c.o): In function `sha_reset':
crypto.c:(.text+0x8a8): undefined reference to `EVP_MD_CTX_reset'
lib/libairplay.a(crypto.c.o): In function `sha_destroy':
crypto.c:(.text+0x924): undefined reference to `EVP_MD_CTX_free'
collect2: error: ld returned 1 exit status
CMakeFiles/rpiplay.dir/build.make:110: recipe for target 'rpiplay' failed
make[2]: *** [rpiplay] Error 1
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/rpiplay.dir/all' failed
make[1]: *** [CMakeFiles/rpiplay.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

FD-, 2019-10-10:
You have an old version of libel.

GhostR111der, 2019-10-10:
Ok, thank you, I installed this) How I can start use this? Then I print rpiplay - nothing happens

GhostR111der, 2019-10-10:
Sorry for my stupidity. I open folder with rpiplay and start service- and I see rpiplay on my iPhone, but when I tried to send video from iphone - my iPhone said "failed to connect"

GhostR111der, 2019-10-10:
Yeeeeeeee) Its work! Thank you for your great job. I just update AVAHI)