FD-/RPiPlay #32: Disconnecting Airplay hangs after playing a video

sumanc, 2019-10-17:
Hi,

I use my iPhone to connect to AirPlay from the control center. After I play any video (for example from yahoo.com) and try to stop screen mirroring from the iPhone, it rpiplay hangs. I narrowed it down to this  function in raop_rtp.c

```
void
raop_rtp_stop(raop_rtp_t *raop_rtp)
{
    assert(raop_rtp);

    /* Check that we are running and thread is not
     * joined (should never be while still running) */
    MUTEX_LOCK(raop_rtp->run_mutex);
    if (!raop_rtp->running || raop_rtp->joined) {
        MUTEX_UNLOCK(raop_rtp->run_mutex);
        return;
    }
    raop_rtp->running = 0;
    MUTEX_UNLOCK(raop_rtp->run_mutex);

    /* Join the thread */
    THREAD_JOIN(raop_rtp->thread);

    if (raop_rtp->csock != -1) closesocket(raop_rtp->csock);
    if (raop_rtp->dsock != -1) closesocket(raop_rtp->dsock);

    /* Flush buffer into initial state */
    raop_buffer_flush(raop_rtp->buffer, -1);

    /* Mark thread as joined */
    MUTEX_LOCK(raop_rtp->run_mutex);
    raop_rtp->joined = 1;
    MUTEX_UNLOCK(raop_rtp->run_mutex);
}
```
it waits on this line THREAD_JOIN(raop_rtp->thread);

If I comment out all the synchronization code as below, it works fine but I am not sure if there are any other impact because of that.

void
raop_rtp_stop(raop_rtp_t *raop_rtp)
{
    assert(raop_rtp);

    /* Check that we are running and thread is not
     * joined (should never be while still running) */
//    MUTEX_LOCK(raop_rtp->run_mutex);
//    if (!raop_rtp->running || raop_rtp->joined) {
//        MUTEX_UNLOCK(raop_rtp->run_mutex);
//        return;
//    }
    raop_rtp->running = 0;
//    MUTEX_UNLOCK(raop_rtp->run_mutex);
//
//    /* Join the thread */
//    THREAD_JOIN(raop_rtp->thread);

    if (raop_rtp->csock != -1) closesocket(raop_rtp->csock);
    if (raop_rtp->dsock != -1) closesocket(raop_rtp->dsock);

    /* Flush buffer into initial state */
    raop_buffer_flush(raop_rtp->buffer, -1);

    /* Mark thread as joined */
//    MUTEX_LOCK(raop_rtp->run_mutex);
//    raop_rtp->joined = 1;
//    MUTEX_UNLOCK(raop_rtp->run_mutex);
}

sumanc, 2019-10-18:
in raop_rtp_thread_udp, it is just looping through         

```
if (raop_rtp_process_events(raop_rtp, NULL)) {
            continue;
}
```
Also, in raop_rtp_mirror_thread, it is looping through;

```
        ret = select(nfds, &rfds, NULL, NULL, &tv);
        if (ret == 0) {
            /* Timeout happened */
            continue;
        }
```

sumanc, 2019-10-18:
In raop_rtp_thread_udp, if I make the following change, it seems to me working;

```
if (raop_rtp_process_events(raop_rtp, NULL)) {
            **break**;
}
```

FD-, 2019-10-18:
There already is a break [there](https://github.com/FD-/RPiPlay/tree/master/lib/raop_rtp.c#L407), so I'm not sure what code you're referring to.

sumanc, 2019-10-18:
Hmm, you are right. Sorry about that. I am not sure how my code is different. I will have to go through the commit history in my fork. Thank you for looking into it and sorry for wasting your time.

FD-, 2019-10-18:
@sumanc Never mind! I know there still are threading issues in there (RPiPlay sometimes cannot be stopped), so I'm thankful for any help with chasing them down!

suman-cherukuri, 2019-10-28:
Hi, you can close this issue. After making sure that my fork is in sync with main, this issue went away.

closed 2019-10-29 by sumanc