FD-/RPiPlay #33: DAC support

kai-lip, 2019-10-18:
Hi, I'm using a DAC (HiFiBerry+) and it looks like it doesn't work with RPiPlay. I would have guessed that RPiPlay uses ALSA as default audio device but it looks like it uses the internal hardware directly. Any chance to add ALSA support?

FD-, 2019-11-12:
Yeah, RPiPlay uses OpenMAX for audio playback, which seems to only support HDMI and analog output. AFAICT, adding ALSA support is not trivial, since audio and video need to have a common clock for audio-video sync. The developers of OMXPlayer seem to have come up with a rather ingenious solution for bridging ALSA to OpenMAX, but it would definitely take some effort to port their code to RPiPlay (because it is in C++ and depends on libav): https://github.com/popcornmix/omxplayer/commit/c84083d15cbd757895025b2cc9fd902b97e3cb8a

TheManchineel, 2019-12-02:
A workaround could be using an HDMI or TOSlink DAC and a splitter. This obviously isn't a solution if you already have or want to use something like a HiFiBerry.

regie-techno, 2020-02-10:
Actually I use omxplayer with the option "-o alsa" which allows to bridge the sound to a hifiberry card. I admit that the sound quality has absolutely nothing to do between the most basic of hifiberry cards and an hdmi output.
What would be the solution to bring this function to RPiPlay? Is it just a programming problem? Or are there hardware blockages?

FD-, 2020-02-10:
@regie-techno Programming problem, as (clearly, IMHO) explained in my comment above.

regie-techno, 2020-02-10:
Yes, I read this part well: so there is above all a problem of clock synchronization between video and audio data. I would like to be able to code this part because it seems necessary to me from an audiophile point of view ...
If not, could we imagine solving this problem just by using "jack" ?

FD-, 2020-02-10:
@regie-techno The most straight-forward solution to the problem would be porting over the omxplayer commit I linked above. This would (AFAICT) retain av sync for free, but could require quite some time to get working, since the code structure of omxplayer and RPiPlay differ considerably.

I'm not really familiar with JACK, but from a quick read through the web page, it doesn't seem better than ALSA for our purpose here, and that still has the av sync issues that could get complicated.

So, I see these possibilities:
1) Port over the OMXPlayer commit linked above that essentially does some magic to allow using ALSA outputs with the OpenMAX API.
2) Integrate an ALSA audio renderer (challenge: keep av synced)
3) Rewrite audio and video renderer components to use something like ffmpeg, gstreamer or similar higher-level media library

FD-, 2020-11-06:
With the recent addition of gstreamer renderers, you might be able to use the gstreamer audio renderer with the OpenMAX video renderer to get DACs working on Raspberry Pi, though this could well lead to a crash (IIRC the OpenMAX renderers are tightly coupled) (and obviously, av sync won't work).

closed 2020-12-19 by pallas

bendschs, 2021-01-16:
whats the state on this issue, are there plans to integrate it?

i would be greatful for any hints how to install latest gstreamer on rpi4 so i can try and help testing.

regie-techno, 2021-01-18:
Hey

To install gstreamer`

sudo apt-get install libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-pulseaudio

After, this is a command line to check if it works
gst-launch-1.0 filesrc location=_/path/of/video/file.mp4_ ! qtdemux name=dmux dmux.video_0 ! queue ! h264parse ! v4l2h264dec ! v4l2convert ! kmssink dmux.audio_0 ! queue ! decodebin ! audioconvert ! autoaudiosink

Have fun !
Here is a link of a interesting tutorial
[](http://www.einarsundgren.se/gstreamer-basic-real-time-streaming-tutorial/)

bendschs, 2021-01-19:
for building on rp4 buster it tells me:

```
- Checking for modules 'gstreamer-1.0>=1.4;gstreamer-sdp-1.0>=1.4;gstreamer-video-1.0>=1.4;gstreamer-app-1.0>=1.4'
--   Package 'gstreamer-sdp-1.0', required by 'virtual:world', not found
--   Package 'gstreamer-video-1.0', required by 'virtual:world', not found
--   Package 'gstreamer-app-1.0', required by 'virtual:world', not found

```

any chance to get these packages?

bendschs, 2021-01-19:
oh ok, needed to install "libgstreamer-plugins-base1.0-dev".

bendschs, 2021-01-19:
i can confirm, that audio ist transfered to my digital pulseaudio device, if i run rpiplay on x11-desktop with gstream. even low latency mode works like a charm. thanks for the support.

cheers
bendsch

bendschs, 2021-01-20:
rpiplay starts a x11 window that does not fill the entire screen. any option to start the window in fullscreen, hide the x11 pointer and the taskbar of the window?

cheers
bendsch

bendschs, 2021-01-26:
i found a way to maximize the window by cli:
wmctrl -r :ACTIVE: -b toggle,fullscreen

the problem now is, that the Rpiplay-window is created only when something is being casted to rpiplay. I am looking for a way to tell the window being casted to immediatly switch to full-screen.

any help?

beastieboy, 2021-07-16:
Unfortunately, that didn't work for me, so I decided to use project: https://github.com/balenalabs/balena-sound/
Works perfectly with Scarlett Solo, but sound only!