FD-/RPiPlay #34: payload_type == 5 frames are not processed

suman-cherukuri, 2019-10-28:
I am seeing few frames with payload_type == 5 in 

```
static THREAD_RETVAL
raop_rtp_mirror_thread(void *arg)
```
we process only of types 0 and 1. As a result, iPhone screens are not fully rendered at the end. Particularly when we scroll horizontally between home screens. 

In one of the home screens, there are few payload_type==55 frames and then one or two payload_type==0 frames come in. I have attached a file with a couple of decrypted frames of type 5 if that helps to debug.
[out5_h264.txt](https://github.com/FD-/RPiPlay/files/3780710/out5_h264.txt)

FD-, 2019-11-05:
Sorry, I don't know what to make out of the attached text file.  I'm seeing no problems when scrolling between home screens on my iPad. 

Do you know what payloads of type 5 contain? 
What do you mean by 'iPhone screens are not fully rendered at the end'? What end?
Can you post a video of what is not fully rendered?
What iOS version are you using?
Why are you referring to payload type 5 in one place and 55 in another?

suman-cherukuri, 2019-11-05:
Sorry, I meant 5 instead of 55. Attached files are supposed to contain the binary data of a couple of sample frames with type = 5.
My iOS version is the latest - 13.1.3
I will also create video and upload it.

FD-, 2019-11-12:
What still isn't clear: What is the effect of not processing type 5 frames? Why are you noticing type 5 frames and others are not (including me)? What's the difference between type 5 frames and normal type 0/1 frames?

closed 2020-02-28 by FD-