FD-/RPiPlay #35: Cmake error

costadelsol06, 2019-11-01:
Hello!
Thanks for this awesome work! I just discovered your tool and I wanted to use it with OSMC on my Pi 3B+. I use the OSMC 1st November update (I have just updated it before to know I was going to need your tool) and after followed your tuto on GitHub, this is the output I have after the cmake command:

root@osmc-Prentis:/home/osmc/RPiPlay/build# cmake ..
CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

Am I missing something? Or is this because I use OSMC?

Thanks a lot!

FD-, 2019-11-01:
The build instructions are for Raspbian. OSMC is not suitable as a build system. It seems like you‘re even missing a C compiler on your system. OSMC also lacks the Broadcom libraries RPiPlay‘s build expects.

FD-, 2019-11-01:
Duplicate of #10

closed 2019-11-01 by FD-