FD-/RPiPlay #36: Not working for MacOS Catalina

blazj, 2019-11-03:
Hi guys!

I've been trying to setup the RPiPlay on Raspberry PI 3, for my home Mac with the new version of macOS Catalina (10.15). I've built the project with no problem, but when I run the `rpiplay` command, my Mac sees the AirPlay server, but it cannot connect. 

Output from my console:
```
./rpiplay -n RPI -a hdmi -b
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
```
If I try to connect over AirPlay, nothing happens, and my Mac gives me a _Could not connect to..._ message. 

Thanks, 
Blaz

FD-, 2019-11-03:
Working fine for me on Catalina. Are you sure you try to connect to RPiPlay via the screen mirroring menu and not the audio AirPlay menu?

blazj, 2019-11-03:
> Working fine for me on Catalina. Are you sure you try to connect to RPiPlay via the screen mirroring menu and not the audio AirPlay menu?

I've tried connecting through AirPlay in Display settings and selecting the RPI Server from the drop-down.

FD-, 2019-11-03:
What does RPiPlay log if you start it with debug logging?

blazj, 2019-11-03:
I found the problem @FD-. 

I was running OpenMax for Android casting in parallel. As soon as I disconnected my Android phone from the local network, AirPlay server started working, and I was able to cast the image to the screen. 

It would be cool to have both Android and Apple devices casting possible in an easy way someday.

Offtopic: Would you recommend starting `rpiplay` as a service at startup?

Respect to everyone doing great work on this project!

FD-, 2019-11-05:
Glad to hear you were able to solve the problem! The problem with starting RPiPlay as a service is that should you have to close it, you'll have to do so without seeing the terminal (since RPiPlay draws on top). That's why I'm using this setup: https://github.com/FD-/RPiPlay/wiki/Starting-RPiPlay-at-system-start

closed 2019-11-05 by FD-