FD-/RPiPlay #37: Output only while casting

baramuse, 2019-11-04:
Hi !

I was wondering if there was any way to be able to launch in background in a way that it would only output an image if there is something casting/connected
I don't know the underlying of using the HDMI output, but what if I want to have my KODI running but RPiPlay taking over when casting ?

Cheers for your awesome work!

FD-, 2019-11-05:
If you start RPiPlay with the -b flag, it will keep a transparent background and only draw when a connection is active. However, I suspect it still won't work with KODI since IIRC both draw through the GPU and there's some layer issue (haven't tried myself).

baramuse, 2019-11-05:
> If you start RPiPlay with the -b flag, it will keep a transparent background and only draw when a connection is active. However, I suspect it still won't work with KODI since IIRC both draw through the GPU and there's some layer issue (haven't tried myself).

I'll test tonight then, thanks :)

baramuse, 2019-11-06:
I works like a charm !
Well, quite... if the casted is full screen then we won't see Kodi behind, but if the casted is an iPhone for example, we'll see kodi backgroup (pretty fun though..)
So maybe we could have a third background option : only draw a black background when the cast started ?

FD-, 2019-11-15:
This should be pretty easy to add, probably as an additional parameter to the -b flag.

Dulatech, 2019-11-19:
> I works like a charm !
> Well, quite... if the casted is full screen then we won't see Kodi behind, but if the casted is an iPhone for example, we'll see kodi backgroup (pretty fun though..)
> So maybe we could have a third background option : only draw a black background when the cast started ?

Hello @baramuse,

Were you able to change the application to allow for the black background to only show up when the cast started?

moda20, 2019-11-19:
@baramuse how do you run it on Kodi ? do you build elsewhere then transfer the executable ? what do you do exactly ?

FD-, 2020-01-04:
Added by #50.

closed 2020-01-04 by FD-

baramuse, 2020-01-05:
> @baramuse how do you run it on Kodi ? do you build elsewhere then transfer the executable ? what do you do exactly ?

I just run RPIPlay in background and then run Kodi on top

moda20, 2020-01-05:
@baramuse ah I thought you created a plugin or something