FD-/RPiPlay #39: Problem with start

Simplyha, 2019-11-12:
Hello,
I've an RPI 4 with 2GB RAM
I've done all that, which was written in the README but after when I start with "./rpiplay -n RPI"
Only this comes: 
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>

Thank you for your support

FD-, 2019-11-12:
I don't see a problem here. Seems to work fine so far. Did you try to connect to it?

Simplyha, 2019-11-12:
Hello it was first a problem with the phone and jet I can connect put an other Problem:
I connect the RPI4 with HDMI and try to load a video but It is only a blackscreen. And console is this:

Accepted IPv4 client on socket 18
Local: 192.168.1.177
Remote: 192.168.1.149
Accepted IPv4 client on socket 20
Local: 192.168.1.177
Remote: 192.168.1.149
raop_rtp starting audio
raop_ntp receive timeout

Simplyha, 2019-11-12:
Update: 
Jet I'm also get metadata.
[image]

FD-, 2019-11-12:
How are you starting the connection on the sender?

Simplyha, 2019-11-13:
Thank you for your Answer now I've a Picture.
But I can't play a Video. 
Whether youtube or a browser video.
And also the audio does not work with "-a hdmi" or "-a analog"

tuxx1112, 2019-11-15:
I'm having the same issue. Does this have something to do with the RPI 4? I'm running on an RPI 4 - 4GB. I can connect but can't seem to get a Youtube video or Spotify song to play.

FD-, 2019-11-15:
Does playing music through the Music app work? Are you aware of #2 and #7?

tuxx1112, 2019-11-15:
> Does playing music through the Music app work? Are you aware of #2 and #7?

I have checked the issues and have been able to connect Airplay through the dedicated button. The Screen Mirroring works just fine - I can even watch recorded videos from my phone - but the audio is not coming along. The output states "raop_rtp starting audio" every time I open a video. The same issue when playing music through the Music app: "raop_rtp starting audio" is printed, the phone thinks it is casting to the Pi, but audio is not coming along. I tried it with and without the '-a hdmi' parameter to force audio through HDMI.

I attached a debug log (first Music app, then recorded video).
[logfile.txt](https://github.com/FD-/RPiPlay/files/3851635/logfile.txt)

FD-, 2019-11-15:
Does audio work if you use low-latency mode (the -l flag)?

tuxx1112, 2019-11-15:
I just tried it without result. I got the same output so didn't upload a new logfile.

Simplyha, 2019-11-17:
I have also tested with the an analog cable but it is the same problem.
An other hint is that: I got metadata if I connect with the "white circle" but no audio and no Picture. And if I connect with the "green circle" I don't get metadata and no audio but a picture.
[image]

FD-, 2019-11-18:
The white circle is for audio only AirPlay, and this protocol is not supported by RPiPlay (use Shairport-sync or Shairplay for that), so it's not supposed to work.
The green circle is for AirPlay mirroring, and this is supposed to work in that the iPhone's screen and audio output should be played on the Raspberry Pi.

Does the [hello_audio sample](https://github.com/raspberrypi/firmware/tree/master/opt/vc/src/hello_pi/hello_audio) (source should be shipped with Raspbian at /opt/vc/src/hello_pi/hello_audio) produce any sound output on your Pi 4?

Simplyha, 2019-11-24:
I've change the HDMI Port and then the audio works but Youtube and Google Videos doesn't work.

FD-, 2019-11-24:
Could you detail how exactly you changed the HDMI port? Do audio and video output on different HDMI ports?

fosple, 2019-12-22:
@FD- Same issue here. 

The rPi4 has 2 HDMI ports (see https://images.vogel.de/vogelonline/bdb/1580200/1580232/39.jpg ). If I use the left port (the one next to the USB-C port) I don't get video -only a black screen. If I use the right port (the one next to the audio-jack) I do get video but no audio.

FD-, 2019-12-22:
@fosple Do you get audio on the port that has no video? IIRC Pi 4 only has audio output on one of the 2 HDMI ports.

fosple, 2019-12-22:
@FD- Currently I don't have my rPi4 with me, so I can't check. Will let you know once I've access to it again.

SaltwaterC, 2020-02-14:
@fosple by default, RPi 4 only outputs sound on HDMI0. You need to manually enable sound output for HDMI1, albeit that's not ideal. More details: raspberrypi/firmware#1243

fosple, 2020-02-14:
@SaltwaterC Thanks a lot!

pallas, 2020-10-06:
This is resolved?

closed 2020-10-06 by pallas