FD-/RPiPlay #40: Unable to connect to "RPiPlay"

supersoaker, 2019-11-23:
First of all, thanks a lot for this repository and your work!

But unfortunately I can't connect my Mac Book (MacBook Pro (Retina, 13-inch, Mid 2014), 10.14.6 (18G1012)) nor my iPhone (iPhone 6 Plus, 12.4.3) to the Apple TV service. E.g.

[image] 

Steps to reproduce:
- flashing and setting up a new Raspberry Pi 3 Model B Version 1.2
- add the raspberry to the wifi
- build the service as described https://github.com/FD-/RPiPlay#building
- execute `./rpiplay -d`
- wait a few minutes (only to make sure the service is running properly)
- try to connect to the Apple TV

The debug output when I try to connect:
```
root@raspberrypi:/home/pi/RPiPlay/build# ./rpiplay -d
> stream info: channel = 2	sample_rate = 44100	frame_size = 480	aot = 39	bitrate = 0
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
```

FD-, 2019-11-23:
From the logs, it seems like RPiPlay never received any connection requests. Are all devices in the same network (without any NAT inbetween)?

supersoaker, 2019-11-23:
Yes, they are in the same Wifi network e.g.:
MacBook - 192.168.1.101
raspberrypi - 192.168.1.104

FD-, 2019-11-23:
I suggest to set up the Pi as a Wifi Hotspot and connect the MacBook/iPhone to that, just in order to factor out anything going on in your network.

supersoaker, 2019-11-23:
ah okay, got the error, I was setting up the raspberry with an ansible playbook, which was concentrating on security and therefore blocking the connection to my devices:
https://github.com/hannseman/ansible-raspbian

closed 2019-11-23 by supersoaker

qqbenq, 2020-04-30:
@supersoaker Could you share a bit more detail about what exactly caused the problem and how did you solve it? I seem to be having a similar issue, and would definitely help a lot :)