FD-/RPiPlay #45: Video in Powerpoint Presentation not playing

doronhorwitz, 2019-12-04:
I plan to use RPiPlay for presentations whereby my iPhone would stream Powerpoint presentations (using Microsoft's Powerpoint iOS app - https://apps.apple.com/us/app/microsoft-powerpoint/id586449534) over Airplay. It's a pretty interesting use case because then I wouldn't need to bring a laptop along to all my presentations and would be able to hold my phone and control the presentation without having to stand behind a laptop.

I've generally gotten it to work, slides and transitions between slides looks great - but playing videos doesn't work. When pushing the play button of a video within a Powerpoint slide the playback doesn't start (the video stays at 00:00).
When not connected over Airplay the the video plays fine on the phone's screen.

I see that there is another issue regarding video streaming (https://github.com/FD-/RPiPlay/issues/7) but it doesn't seem connected because when streaming a video from Powerpoint the video has to stream from the iPhone (i.e not for a URL).

My setup is:
1) iPhone 8
2) iOS 13.1.3
3) Raspberry Pi 3 Model B+
4) Raspbian Buster Lite (https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-09-30/2019-09-26-raspbian-buster-lite.zip)

I'd also like to give you whatever other info you'd need (logs etc.). If so, some guidance would be needed to get that info.

FD-, 2019-12-04:
The problem might well be linked to issue #7. The iPhone itself can acts as a web server for the video.

Does the video play when you connect to a real AppleTV via AirPlay mirroring?

doronhorwitz, 2019-12-14:
Finally had a chance to test what you asked.
On a real AppleTV the videos in Powerpoint worked fine.

This then made me suspect that the video should work fine in Powerpoint.

So I went to test again. But this time I actually prepared the presentation before turning on AirPlay. When inserting videos, the Powerpoint iOS app compresses them before inserting into the presentation.
I think this helped and when I streamed the presentation through RPiPlay this time the video playback worked.

I guess trying to play uncompressed video was the problem. Sorry if I created confusion.

Unfortunately now the next problem I'm experiencing is one of stability, as RPiPlay tends to sometimes crash after 2 or 3 minutes of using Powerpoint and playing videos. The image gets stuck on a single frame and stops showing what's actually on the iPhone. I have to completely reboot the RaspberryPI to get it working again. Would you prefer I opened up another issue with my finds and close this current one?

FD-, 2019-12-14:
Yeah, please open a separate issue for that.

closed 2019-12-14 by doronhorwitz