FD-/RPiPlay #46: Error in setup phase

andreabasile, 2019-12-06:
i follow the readme.md file for first setup but i receive this error

Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libcrypto.so (found version "1.1.1d")
-- Configuring done
CMake Error at renderers/CMakeLists.txt:26 (add_library):
  No SOURCES given to target: ilclient

-- Build files have been written to: /home/pirate/RPiPlay/build

FD-, 2019-12-06:
You're not using Raspbian, are you?

andreabasile, 2019-12-06:
more or less.
I try to intall on hypriot that was based on Raspbian

FD-, 2019-12-06:
Well, apparently it doesn't include the OpenMAX ilclient library.

FD-, 2019-12-06:
Duplicate of #10

closed 2019-12-06 by FD-

andreabasile, 2019-12-07:
I was installed Raspian and the error is change.
Now when launch make command i see this error 

`[  2%] Building C object renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/h264_stream.c.o
during RTL pass: mach
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c: In function 'read_scaling_list':
/home/pi/RPiPlay/renderers/h264-bitstream/h264_stream.c:491:1: internal compiler error: Segmentation fault
 }
 ^
0xb6b4611f ???
	../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <file:///usr/share/doc/gcc-8/README.Bugs> for instructions.
make[2]: *** [renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/build.make:102: renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/h264_stream.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:575: renderers/h264-bitstream/CMakeFiles/h264-bitstream.dir/all] Error 2
make: *** [Makefile:84: all] Error 2`

What i  did i do wrong now?

Thanks

FD-, 2019-12-07:
Wow, I have no idea what might make a compiler crash. As always, it might have been a problem with your power supply (compiling is CPU-intensive and a low-power situation is known to potentially cause memory corruption). If that's not the case, I suspect you're _not_ running a clean fresh install of the newest Raspbian image.