FD-/RPiPlay #47: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

moda20, 2019-12-09:
I have an error when running cMake, 
I am running this on a docker container of stretch. here is the console output :

```
You have called ADD_LIBRARY for library ilclient without any source files. This typically indicates a problem with your CMakeLists.txt file
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
BCM_HOST
    linked by target "ilclient" in directory /home/RPPLAY/RPiPlay/renderers
BRCM_EGL
    linked by target "ilclient" in directory /home/RPPLAY/RPiPlay/renderers
BRCM_GLES_V2
    linked by target "ilclient" in directory /home/RPPLAY/RPiPlay/renderers
OPENMAXIL
    linked by target "ilclient" in directory /home/RPPLAY/RPiPlay/renderers
VCHIQ_ARM
    linked by target "ilclient" in directory /home/RPPLAY/RPiPlay/renderers
VCOS
    linked by target "ilclient" in directory /home/RPPLAY/RPiPlay/renderers

-- Configuring incomplete, errors occurred!
See also "/home/RPPLAY/RPiPlay/build/CMakeFiles/CMakeOutput.log".
See also "/home/RPPLAY/RPiPlay/build/CMakeFiles/CMakeError.log".
``` 
 
doesn't Iclient come with raspbian integrated ?

FD-, 2019-12-09:
Does it work outside of Docker? I bet it does.

moda20, 2019-12-09:
just a question, is the manual compiling way totally necessary for development ? or is it something else ?

FD-, 2019-12-09:
I'm afraid I don't understand your question. What 'manual compiling way' are you talking about?

moda20, 2019-12-09:
i am asking why is there no release binaries that are already compiled, so everybody who uses the library wouldn't have to compile it using Cmake and make and doesn't need to have a raspbian machine to do so.

FD-, 2019-12-09:
IIRC, the Broadcom libraries are linked dynamically, so they're needed at runtime, which likely means RPiPlay will only run on Raspbian anyway. Most of the open source Raspbian programs are only available in source code form, and most people wouldn't feel confident running a binary that someone else compiled (and may or may not do what's claimed). If you're using a Raspberry Pi, education is part of the experience, so consider this part of your Linux education.

moda20, 2019-12-09:
Okey thanks, I'll close the issue

closed 2019-12-09 by moda20