FD-/RPiPlay #48: RTSP H264 feed to ffmpeg

saket424, 2019-12-12:
Thanks for this amazing opensource project. I was wondering if instead of rendering the feed locally on the pi, the h264 rtsp video feed that is negotiated during the mirroring is available to something like ffmpeg to send to a remote RTMP/RTSP server .

Thoughts?

FD-, 2019-12-12:
Not at the moment, but it should be easy to do, as sketched in #24.

saket424, 2019-12-12:
If this https://nto.github.io/AirPlay.html#screenmirroring is still accurate, video screen mirroring is a TCP connection sending the H264 payload directly over a TCP connection and not via RTSP.

I will explore it in the coming days and see how we may be able to get the payload sent over to ffmpeg. Thanks

FD-, 2019-12-12:
Sorry, I thought you wanted to just pipe the raw H264 stream out of RPiPlay. I'm not sure how you'd convert that into an RTSP stream, but I assume there must be some tool capable of doing that. The AirPlay protocol itself doesn't involve an RTSP stream that could be simply rerouted.

FD-, 2020-01-02:
Duplicate of #24

closed 2020-01-02 by FD-