FD-/RPiPlay #49: crash under 4K resolution

lightingghost, 2019-12-15:
First I would like to thank you for making this great project!

The server works great under 1080p resolution. When I tried to set the output of my raspberry Pi 3 to be 4K, the raspbian gui runs fine, and the rpiplay server can start. But when I tried to cast the screen to the RPiPlay, the system crashed.

Is it because 4K is not support on rpi 3B, or RPiPlay does not support 4K mirroring?

Thanks

FD-, 2019-12-15:
Could you be a bit more detailed? How do you try to set the output of the Pi 3 to 4K? What device‘s screen are you trying to cast? How did the system crash? Any crash log?

lightingghost, 2019-12-16:
Thank you @FD- for helping. I am answering the questions below:

> How do you try to set the output of the Pi 3 to 4K?

I am following this [guide](http://support.yodeck.com/knowledgebase/articles/1912408-define-an-advanced-custom-and-a-4k-resolution) to change the configs.txt as following:

```
gpu_mem_1024=256
framebuffer_width=3840
framebuffer_height=2160
max_framebuffer_width=3840
max_framebuffer_height=2160
hdmi_ignore_edid=0xa5000080
hdmi_drive=2
hdmi_pixel_freq_limit=400000000
hdmi_group=2
hdmi_mode=87
hdmi_timings=3840 1 48 32 80 2160 1 3 5 54 0 0 0 25 0 220430000 3
```

> What device‘s screen are you trying to cast?

I have tried to cast from a 2019 MBP15 and an iPhone XR. Both caused the crash.

> How did the system crash? 

The monitor connected to the RPi 3 shows a pop up window saying "no signal", therefore at least the screen output is gone. I have no idea of how to verify whether the OS is still running.

> Any crash log?

Where shall I find the log?

FD-, 2019-12-16:
1. Anything beyond 1080p is clearly out of specs for Pi earlier than Pi4.
2. Even the Guide you linked warns that video decoding doesn‘t work at that resolution.
3. What you‘re seeing could even be the monitor crashing due to uncommon framerate.
4. AFAICT, AirPlay doesn‘t support 4K resolution anyway.

closed 2019-12-16 by FD-