FD-/RPiPlay #52: Great job. Everything just worked.

danielbalog, 2020-01-02:
I don't have an issue at all, but I didn't know what the best way to thank you was.

I just built this on my Rasberry Pi 3B+. Installation was easy. Hooked it up to my TV and it just worked.

I tried with my iPhone XR and my MacBook Pro, and both just worked extremely fast with barely any lag at all. I could run both video and audio. Honestly blew my mind.

I'm so glad I discovered this. Keep up the good work!

( Feel free to close this as resolved ;-) )

patientzero, 2020-01-04:
Same as my @danielbalog.
Just thanks a lot. After battling with other solutions the last couple of years, this just worked!
So Thank you!

alistarle, 2020-03-22:
Same for me, this project is setup on my pi in less than 5 minutes and works like a charm, good work dude !

iDoka, 2020-03-26:
What I could say about this project?
It just worked!!! Amazing!
Build instruction is clear and fine!
@FD-  Many thanks for this repo!

rerpha, 2020-04-03:
+1, Have used this on both a RPi model 2B and a Zero W 1.1, works really nicely on both!

kirschem, 2020-05-11:
+++
Also worked great for me, many thanks! Sound and video from my iPhonxe XS worked out of the box with Model 3.

I just made a simple script to automate the installation and add the server to autostart: 
https://github.com/kirschem/install-rpiplay

pallas, 2020-10-06:
Glad to hear it!

closed 2020-10-06 by pallas