FD-/RPiPlay #53: Crashes during video playback (likely audio related)

philippjbauer, 2020-01-05:
First off, THANK YOU for making this!

I experiences instabilities that make the airplay server pretty much unusable until I started using the low latency mode.

Here's my setup and steps to reproduce the issue:

**Client:**
MacBook Pro 2015, macOS Mojave

**Server:**
Raspberry Pi 3, Latest Raspian OS (Buster Sep 2019, Clean Install)
Connected to DELL P2715Q @1080p/30Hz via HDMI

Started RPiPlay without options and with forcing audio to HDMI.

**Steps to reproduce:**
1) Start RPiPlay
2) Extend macOS Display to AirPlay
3) Open Firefox window on extended display
4) Open any video on YouTube
5a) First video plays without interruption, next video (autoplay) has no audio and crashes server
or
5b) First video plays, skip forward in video, audio stops and crashes server

I can't exit with `ctrl+c`, so I have no way to check the debug output when I enable it to give you any log output. Please advise on how I could accomplish that.

**Current Solution:**
I was curious if the low latency mode would change anything and it works very well. No crashes and I can skip through a YouTube video. I can connect and reconnect which didn't work previously either.

Is there a way to log into a file so I can retrieve the logs after a crash? My assumption right now is that the synchronization of audio/video crashes and pulls everything down with it.

corburn, 2020-01-19:
RPiPlay version: 05ed7931529d240fb0bea9d4a858aad229723a01

I think have the same problem

```
./rpiplay -b auto -l
raop_rtp_mirror starting mirroring
raop_rtp_mirror starting audio
raop_rtp_mirror starting audio
raop_rtp_mirror receive timeout
free(): double free detected in tcache 2
Aborted
```

essboyer, 2020-01-20:
Without seeing your debug output, I'd hazard a guess that it's network related. 
Are both your client and server devices connected via ethernet?  Is the port you're connecting over open for both TCP and UDP?

pallas, 2020-05-05:
@corburn, can you please try again with the latest version? I bumped FDK-AAC and several other libraries recently.

pallas, 2020-10-06:
Without more detailed debugging information, this isn't possible to resolve. I'm going to hope that upstream fixes helped.

closed 2020-10-06 by pallas