FD-/RPiPlay #54: Rotate image 90 degrees

wimrigo, 2020-01-09:
My screen is standing up vertically but when I cast it shows the image horizontally. Is there a way to flip the image?

rcarmo, 2020-01-10:
Do you have `display_rotate` in your `config.txt`? That is how the Pi knows how to rotate conventional text and graphics.

wimrigo, 2020-01-11:
display_rotate isn't working on my PI4. I've read that it isn't supported for PI4.
I rotate my screen via 'preferences'

rcarmo, 2020-01-11:
Ah. So you’re doing it at the X11/RandR level most likely, not the framebuffer/hardware level.

wimrigo, 2020-01-14:
Like they mention in the link below it is not possible due to the new video driver they use:(https://pimylifeup.com/raspberry-pi-rotate-screen/)

FD-, 2020-01-14:
This is probably trivial to do, just by setting the [OMX_IndexConfigCommonRotate](http://www.jvcref.com/files/PI/documentation/ilcomponents/video_render.html) config on the OpenMAX renderer component. I'd suggest adding a -r {degree} argument to the cmdline.

StevenHickson, 2020-01-22:
Any more insight into how to do this?
I tried adding the following to the file *renderers/video_renderer_rpi.c* and it just fails to initialize:
```
    // Setup rotation
    OMX_CONFIG_ROTATIONTYPE rotation;
    memset(&rotation, 0, sizeof(OMX_CONFIG_ROTATIONTYPE));
    rotation.nRotation = 90;
    rotation.nPortIndex = 91;
    if (OMX_SetParameter(ilclient_get_handle(renderer->video_renderer), OMX_IndexConfigCommonRotate,
                         &rotation) != OMX_ErrorNone) {
        video_renderer_destroy_decoder(renderer);
        return -15;
    }
```

I also tried adding the transform to the DisplayRegion per the docs with the line:
```
display_region.transform = OMX_DISPLAY_ROT90;
```
That seems like it should work based off of this documentation:
https://github.com/adafruit/rpi-firmware/blob/master/vc/sdk/opt/vc/include/IL/OMX_Broadcom.h
but it just does nothing and keeps the normal rotation.

StevenHickson, 2020-01-23:
Actually I figured this out and submitted a PR#58 for it.

wimrigo, 2020-01-30:
@StevenHickson What do you mean with a PR#58?

StevenHickson, 2020-01-31:
Pull Request.
https://github.com/FD-/RPiPlay/pull/58
You can change the number on line 279 rotation.nRotation = -90 to suit your rotation. The way to do this cleanly would probably be to allow the rotation number as the command line flag but I'm waiting to hear from the author before doing anything like that.

You should be able to grab that pull request and run a rotated version. It is working perfectly well for me at home.

wimrigo, 2020-02-09:
I copied those 4 files and started rpiplay this way: ./rpiplay -n cast -r
Unfortunately when I cast there is no rotation. Am I doing something wrong?

wimrigo, 2020-02-09:
@StevenHickson I copied those 4 files and started rpiplay this way: ./rpiplay -n cast -r
Unfortunately when I cast there is no rotation. Am I doing something wrong?

StevenHickson, 2020-02-18:
You need to recompile. An updated #67 let's you chose the rotation amount. You can also pull that specific patch or pull request and compile.

FD-, 2020-02-28:
Merged now.

closed 2020-02-28 by FD-