FD-/RPiPlay #56: Connecting with an active connection will freeze

vicary, 2020-01-20:
This worked well if only worked with one single device, given that it cleanly disconnects before next connection.

Screen freezing begins when some other devices try to connect with an active one already in place, or an iPhone fiddles with video casting where it “switches channels” on the fly. Sometimes it recovers, most of the time it requires manually killing the process.

I know a little bit of C, but not quite enough to start debugging this I believe.

Where shall I start?

FD-, 2020-01-20:
What do you mean by 'cleanly disconnects before next connection'? What next connection? As for debugging, it's always a good start to provide a debug log (use the -d flag).

vicary, 2020-01-20:
> What do you mean by 'cleanly disconnects before next connection'? What next connection?
Sometimes it can’t reconnect again after connection timeout, more likely to happen when I cast my phone screen via GSM VPN for remote conferences. Was suspecting it comes from the same underlying cause with the “other device” one.

I’ll send a debug log when I have my laptop tomorrow.

closed 2020-02-28 by FD-