FD-/RPiPlay #59: How to do?

marmod, 2020-01-31:
how to do RPIPlay was enabled all the time on the system instead of starting it using the command

sequentiality, 2020-02-02:
@marmod you can create systemd service. Here's mine:
```
❯ cat /etc/systemd/system/rpiplay.service
[Unit]
Description=An open-source AirPlay mirroring server for the Raspberry Pi
Documentation=https://github.com/FD-/RPiPlay
After=network.target

[Service]
Type=simple
Restart=always
Environment=NAME="Raspberry Pi"
Environment=AUDIO=hdmi
Environment=BACKGROUND=auto
ExecStart=/usr/local/bin/rpiplay -b $BACKGROUND -n ${NAME} -a $AUDIO

[Install]
WantedBy=multi-user.target
❯ sudo systemctl status rpiplay.service
● rpiplay.service - An open-source AirPlay mirroring server for the Raspberry Pi
   Loaded: loaded (/etc/systemd/system/rpiplay.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-02-02 11:57:42 EET; 6min ago
     Docs: https://github.com/FD-/RPiPlay
 Main PID: 4014 (rpiplay)
    Tasks: 10 (limit: 4915)
   Memory: 21.3M
   CGroup: /system.slice/rpiplay.service
           └─4014 /usr/local/bin/rpiplay -b auto -n Raspberry Pi -a hdmi

Feb 02 11:57:42 raspberrypi systemd[1]: Started An open-source AirPlay mirroring server for the Raspberry Pi.
```

FD-, 2020-02-02:
Alternatively, have a look at the [Wiki](https://github.com/FD-/RPiPlay/wiki).

closed 2020-02-02 by FD-