FD-/RPiPlay #60: OSX catalina vs ios 13.3.1

2-face, 2020-02-03:
Not sure what is related to but mirroring from Catalina (Mac pro 13 touchbar) to RPI zero is just pretty much real-time & smooth. However mirroring from Ios 13.3.1 (Iphone 8) is just sluggish, delayed and pretty much unusable.

Any idea what it could be related to ?

FD-, 2020-02-04:
Is the iPhone probably connected via Wifi and the Mac(Book?) Pro via ethernet cable? For me, it's working fine from an iPad running iOS 13.3 (wired Internet connection).

caesar753, 2020-02-06:
For me mirroring from ipad 13.1 and raspberry pi 3, both connected through wifi, goes very smooth, but not working at all trying to mirror videos (youtube, netflix, amazon prime)

FD-, 2020-02-06:
@caesar753 See issue #7

caesar753, 2020-02-06:
ok, thanks, video cast is not implemented (yet)

LeXuanKhanh, 2020-02-10:
hi @2-face , can you tell me what rasberry pi you use ? cause i wanna buy 1 and setup this with my hackintosh currently running catalina but idk what pi specs to mirror the screen at native resolution (1920x1080) smoothly

FD-, 2020-02-10:
@LeXuanKhanh For best performance with both audio and video, use a Pi 3B+ or newer. Probably a Pi 2 would do as well, but I don't have one to test with. On a Pi Zero (and probably the slower Pi 1 as well), video + audio can work smoothly, but only in ideal situations.

2-face, 2020-02-23:
I use RPI zero - looks that after moving it and connecting my TV it works ok. Well, IOS still is not ideal i.e. when connected it is fine but it might be that iPhone disconnects after checking few videos/photos, but not always, it looks that after some idle time it just disconnects. OS X works like a charm, interestingly My Sony NEX 6 can be connected to it rpiplay server and it works great as well. I wasn't aware that Sony camera is capable of utilising Airplay for mirroring.

FD-, 2020-02-28:
I highly doubt the Sony supports AirPlay mirroring.

closed 2020-02-28 by FD-