FD-/RPiPlay #65: audio receiving issues

krssubbu, 2020-02-14:
When I do a screen cast from iPhone YouTube app to TV via RPiPlay, I dont see video or audio and  
I get the following error
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.0.30
Remote: 192.168.0.14
Accepted IPv4 client on socket 17
Local: 192.168.0.30
Remote: 192.168.0.14
raop_rtp starting audio
raop_ntp receive timeout

When I do screen mirroring, I see the images or screen on TV but no audio. 

I am using an Apple iPhone 12.4.5

jotoid, 2020-02-15:
Do you get audio with option -l ? For some reason audio works only if I flag -l

krssubbu, 2020-02-15:
Thanks for quick reply I will try this and sorry I am new to this, can you give me the full command please? I guess the option you are saying is for screen mirror where I see only video and no audio

when casting YouTube to TV, I don’t get audio and video, only black screen with error saying something like ntpd not received (exact error posted on subject line) and get black screen and won’t be able to see anything unless I reboot pi

jotoid, 2020-02-15:
> Thanks for quick reply I will try this and sorry I am new to this, can you give me the full command please? I guess the option you are saying is for screen mirror where I see only video and no audio [...]

What is the command you use to run rpiplay?

krssubbu, 2020-02-15:
I do ./RPiPlay but even if I use a parameter

./RPiPlay -a hdmi I don’t get the sound

FD-, 2020-02-15:
YouTube is not supposed to work, as you would have found out if you browsed through other open issues. Regarding audio, you forgot to mention what Pi you are using and whether audio from other programs is working (audible).

jotoid, 2020-02-15:
in my case audio only works when I do -l (both with hdmi and analog options) with the known caveats re sync. I use a pi 4 b+ with buster and 4.19.97-v8+

krssubbu, 2020-02-17:
Sorry for late reply

I didn’t realize YouTube has issues

It’s a old pi and I brought it in 2013 think it’s Pi B model

krssubbu, 2020-02-17:
Sorry what option you use to get audio?

I used ./RPiPlay -l hdmi. - I think I tried it now (the volume was low) but when I do the screen sharing via YouTube is an issue

closed 2020-02-28 by FD-