FD-/RPiPlay #70: can't see the screen :P

nickdebesten, 2020-02-26:
if i try to monitor my screen in the console i see this: 
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=rpiplay>
Accepted IPv4 client on socket 15
Local: 192.168.1.108
Remote: 192.168.1.102
raop_ntp receive timeout
Accepted IPv4 client on socket 17
Local: 192.168.1.108
Remote: 192.168.1.102
raop_rtp_mirror starting mirroring
but the screen of the raspberry pi is still black
i'm using a iphone 6s with IOS 13.3.1

nickdebesten, 2020-02-26:
i also tried music and got this 
Accepted IPv4 client on socket 15
Local: 192.168.1.108
Remote: 192.168.1.102
Accepted IPv4 client on socket 17
Local: 192.168.1.108
Remote: 192.168.1.102
raop_ntp receive timeout
raop_rtp starting audio
aacDecoder_DecodeFrame error : 0x400c
Got image data of 74222 bytes
aacDecoder_DecodeFrame error : 0x4002
aacDecoder_DecodeFrame error : 0x4002
aacDecoder_DecodeFrame error : 0x4002
aacDecoder_DecodeFrame error : 0x4002
aacDecoder_DecodeFrame error : 0x4002
aacDecoder_DecodeFrame error : 0x4002
Got metadata of 185 bytes
Got metadata of 124 bytes
Got image data of 74222 bytes
Got metadata of 185 bytes
Got metadata of 124 bytes
i hear nothing

FD-, 2020-02-28:
1. Likely a duplicate of #5 
2. Not supposed to work (#2)

closed 2020-02-28 by FD-