FD-/RPiPlay #71: Crackling audio over 3.5mm analog jack

jotoid, 2020-02-28:
Hi thanks for putting this together. It's been really useful.

I'm using a raspberry pi 4 with a fresh install. When using the analog jack with rpiplay they audio quality is bad with crackling/static noise. 
This is not the case when running other things (retropie, shairport-sync) through the analog jack. Audio is fine over HDMI in all use cases.  Here's my debug 
` stream info: channel = 2	sample_rate = 44100	frame_size = 480	aot = 39	bitrate = 0
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.0.37
Remote: 192.168.0.13
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = -1286895404
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.0.37/12592026146374895611
DACP-ID: 5FB82C59F5FCD98
Active-Remote: 995917004
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 64949
raop_ntp parse remote ip = 192.168.0.13
raop_ntp starting time
raop_rtp parse remote ip = 192.168.0.13
raop_ntp send_len = 32
raop_rtp_mirror parse remote ip = 192.168.0.13
eport = 40521, tport = 55350
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1582832718049919
Accepted IPv4 client on socket 17
Local: 192.168.0.37
Remote: 192.168.0.13
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = -1286895404
httpd receiving on socket 15
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.0.37/12592026146374895611
httpd receiving on socket 15
conn_request
Handling request RECORD with URL rtsp://192.168.0.37/12592026146374895611
raop_handler_record
httpd receiving on socket 15
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.0.37/12592026146374895611
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.0.37/12592026146374895611
[... 92 lines cut ...]
Got h264 data of 42 bytes
Injecting max_dec_frame_buffering
Video delay is 1582897925185321
raop_rtp_mirror video ntp = 1582897925193602, now = 1582897925185457, latency = -8145
Got h264 data of 763 bytes
Video delay is -7440
raop_rtp_mirror video ntp = 1582897925226935, now = 1582897925186300, latency = -40635
Got h264 data of 964 bytes
Video delay is -39933
raop_rtp_mirror video ntp = 1582897925260268, now = 1582897925194986, latency = -65282
Got h264 data of 851 bytes
Video delay is -64563
raop_rtp_mirror video ntp = 1582897925293600, now = 1582897925230044, latency = -63556
Got h264 data of 1390 bytes
Video delay is -62273
raop_rtp_mirror video ntp = 1582897925326933, now = 1582897925259088, latency = -67845
Got h264 data of 989 bytes
Port settings changed!!
Video pipeline delay is 6 frames or 75862 us
Video delay is -56727
raop_rtp audio: ntp = 1582897925259775, now = 1582897925270819, latency=11044, rtp=2090128619
raop_rtp audio: ntp = 1582897925270659, now = 1582897925271038, latency=379, rtp=2090129099
raop_rtp audio: ntp = 1582897925281543, now = 1582897925271386, latency=-10157, rtp=2090129579

`

FD-, 2020-02-28:
Are you sure the setup is the same for hdmi and analog? Crackling can be caused by the Pi being busy with other programs or by a bad WiFi connection. Other than that, I‘m afraid there‘s nothin RPiPlay could do about the problem. HDMI and analog use the same code.

closed 2020-02-28 by FD-

MrRiddler86, 2020-06-17:
Hi all!

I am trying to run exactly the same setup (raspberry pi 4b with 4GB RAM) as @jotoid and obtain exactly the same problem.
I could also verify, that, e.g. shairport-sync, works fine on analog output.

So, this seems to be somewhat reproducible on raspberry pi 4.

Best regards

ivma, 2020-08-25:
Same for me

FD-, 2020-08-25:
Does using low-latency mode fix the problem?

ivma, 2020-08-31:
Thanks for the quick response. No unfortunately

jotoid, 2020-09-25:
> Are you sure the setup is the same for hdmi and analog? Crackling can be caused by the Pi being busy with other programs or by a bad WiFi connection. Other than that, I‘m afraid there‘s nothin RPiPlay could do about the problem. HDMI and analog use the same code.

By this logic I should have the same issue with HDMI audio which was not the case.