FD-/RPiPlay #73: P4 Crashes on Screen rotation

mrwrighty, 2020-02-28:
Have done a clean build on P4 and all works well in standard display, but as soon as I add rotation it crashes the Pi4 completely.

FD-, 2020-02-28:
Did you try different rotation values? Which ones?

mrwrighty, 2020-02-28:
Tried 90 180 and 270

mrwrighty, 2020-03-02:
Hi cannot get a log file or output as RPiPlay crashes as soon as the connection is made.

have tried RPiPlay -r 90 -l -d and also 270 rotation. Also switch audio output so see if that helped, but no change. Tried to capture debug output to file with rpiplay -r 90 -d > rpiplay.log, but nothing captured. Having the option to create a log file would be a great add on.

FD-, 2020-03-02:
Since the same code works fine older Pis, I suspect this is actually a bug in the Pi4 (or the OS). That's why having a kernel log or similar crash report for the Pi itself would be useful. I don't think an RPiPlay log would contain any useful information for debugging this.

mrwrighty, 2020-03-02:
OK kern.log output from starting rpiplay to the crash.

`Mar  2 11:27:19 raspberrypi kernel: [    3.286291] rpivid-mem feb00000.hevc-decoder: rpivid-hevcmem initialised: Registers at 0xfeb00000 length 0x0000ffff
Mar  2 11:27:19 raspberrypi kernel: [    3.286922] rpivid-mem feb10000.rpivid-local-intc: rpivid-intcmem initialised: Registers at 0xfeb10000 length 0x00000fff
Mar  2 11:27:19 raspberrypi kernel: [    3.292385] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
Mar  2 11:27:19 raspberrypi kernel: [    3.293261] media: Linux media interface: v0.10
Mar  2 11:27:19 raspberrypi kernel: [    3.296617] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
Mar  2 11:27:19 raspberrypi kernel: [    3.296638] [vc_sm_connected_init]: start
Mar  2 11:27:19 raspberrypi kernel: [    3.299453] rpivid-mem feb20000.h264-decoder: rpivid-h264mem initialised: Registers at 0xfeb20000 length 0x0000ffff
Mar  2 11:27:19 raspberrypi kernel: [    3.300078] rpivid-mem feb30000.vp9-decoder: rpivid-vp9mem initialised: Registers at 0xfeb30000 length 0x0000ffff
Mar  2 11:27:19 raspberrypi kernel: [    3.307394] [vc_sm_connected_init]: installed successfully
Mar  2 11:27:19 raspberrypi kernel: [    3.314661] videodev: Linux video capture interface: v2.00
Mar  2 11:27:19 raspberrypi kernel: [    3.350069] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Mar  2 11:27:19 raspberrypi kernel: [    3.352092] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
Mar  2 11:27:19 raspberrypi kernel: [    3.359926] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
Mar  2 11:27:19 raspberrypi kernel: [    3.363051] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
Mar  2 11:27:19 raspberrypi kernel: [    3.379469] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
Mar  2 11:27:19 raspberrypi kernel: [    3.379509] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
Mar  2 11:27:19 raspberrypi kernel: [    3.388061] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
Mar  2 11:27:19 raspberrypi kernel: [    3.388097] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
Mar  2 11:27:19 raspberrypi kernel: [    3.393959] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
Mar  2 11:27:19 raspberrypi kernel: [    3.394001] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
Mar  2 11:27:19 raspberrypi kernel: [    3.599435] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
Mar  2 11:27:19 raspberrypi kernel: [    3.619342] bcm2835_audio soc:audio: card created with 8 channels
Mar  2 11:27:19 raspberrypi kernel: [    3.621870] cfg80211: Loading compiled-in X.509 certificates for regulatory database
Mar  2 11:27:19 raspberrypi kernel: [    3.760847] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
Mar  2 11:27:19 raspberrypi kernel: [    3.846879] brcmfmac: F1 signature read @0x18000000=0x15264345
Mar  2 11:27:19 raspberrypi kernel: [    3.869380] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Mar  2 11:27:19 raspberrypi kernel: [    3.870215] usbcore: registered new interface driver brcmfmac
Mar  2 11:27:19 raspberrypi kernel: [    3.873254] vc4-drm soc:gpu: bound fe600000.firmwarekms (ops vc4_fkms_ops [vc4])
Mar  2 11:27:19 raspberrypi kernel: [    3.873274] checking generic (3e402000 7f8000) vs hw (0 ffffffffffffffff)
Mar  2 11:27:19 raspberrypi kernel: [    3.873284] fb: switching to vc4drmfb from simple
Mar  2 11:27:19 raspberrypi kernel: [    3.873553] Console: switching to colour dummy device 80x30
Mar  2 11:27:19 raspberrypi kernel: [    3.884011] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
Mar  2 11:27:19 raspberrypi kernel: [    3.884026] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
Mar  2 11:27:19 raspberrypi kernel: [    3.884036] [drm] No driver support for vblank timestamp query.
Mar  2 11:27:19 raspberrypi kernel: [    3.884045] [drm] Setting vblank_disable_immediate to false because get_vblank_timestamp == NULL
Mar  2 11:27:19 raspberrypi kernel: [    3.960770] [drm] Initialized v3d 1.0.0 20180419 for fec00000.v3d on minor 1
Mar  2 11:27:19 raspberrypi kernel: [    4.102400] Console: switching to colour frame buffer device 240x67
Mar  2 11:27:19 raspberrypi kernel: [    4.143089] vc4-drm soc:gpu: fb0: DRM emulated frame buffer device
Mar  2 11:27:19 raspberrypi kernel: [    4.149729] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Mar  2 11:27:19 raspberrypi kernel: [    4.170424] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
Mar  2 11:27:19 raspberrypi kernel: [    5.680496] uart-pl011 fe201000.serial: no DMA platform data
Mar  2 11:27:19 raspberrypi kernel: [    5.707596] random: crng init done
Mar  2 11:27:19 raspberrypi kernel: [    5.707614] random: 7 urandom warning(s) missed due to ratelimiting
Mar  2 11:27:19 raspberrypi kernel: [    5.910237] 8021q: 802.1Q VLAN Support v1.8
Mar  2 11:27:19 raspberrypi kernel: [    6.121523] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
Mar  2 11:27:20 raspberrypi kernel: [    6.189848] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[... cut ...]
Mar  2 11:27:26 raspberrypi kernel: [   12.618319] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Mar  2 11:27:26 raspberrypi kernel: [   12.618326] Bluetooth: BNEP filters: protocol multicast
Mar  2 11:27:26 raspberrypi kernel: [   12.618337] Bluetooth: BNEP socket layer initialized
Mar  2 11:27:26 raspberrypi kernel: [   12.671696] Bluetooth: RFCOMM TTY layer initialized
Mar  2 11:27:26 raspberrypi kernel: [   12.671709] Bluetooth: RFCOMM socket layer initialized
Mar  2 11:27:26 raspberrypi kernel: [   12.671727] Bluetooth: RFCOMM ver 1.11
`

keot, 2020-03-21:
This does indeed lock up the entire Raspberry Pi 4. It renders the first frame (albeit slightly garbled) then locks up.

```
Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
Current Operating System: Linux james-block 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M video=HDMI-A-1:1920x1080M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 smsc95xx.macaddr=DC:A6:32:0C:61:8E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 05 June 2019  12:49:54PM
xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
```
Works fine with no rotation. Nothing in the log-files (`/var/log/*`) of interest.

Debug output:
```
pi@james-block:~/dev/RPiPlay/build $ ./rpiplay -r 270 -d
> stream info: channel = 2	sample_rate = 44100	frame_size = 480	aot = 39	bitrate = 0
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 15
Local: 192.168.0.125
Remote: 192.168.0.106
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = -1287677740
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.0.125/14515932154768477585
DACP-ID: 3DFC8E540135ECD1
Active-Remote: 1219664168
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 52957
raop_ntp parse remote ip = 192.168.0.106
raop_ntp starting time
raop_rtp parse remote ip = 192.168.0.106
raop_rtp_mirror parse remote ip = 192.168.0.106
eport = 44349, tport = 48030
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1582779084790647
Accepted IPv4 client on socket 17
Local: 192.168.0.125
Remote: 192.168.0.106
httpd receiving on socket 15
conn_request
Handling request GET with URL /info
INFO len = -1287677740
httpd receiving on socket 15
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.0.125/14515932154768477585
httpd receiving on socket 15
conn_request
Handling request RECORD with URL rtsp://192.168.0.125/14515932154768477585
raop_handler_record
httpd receiving on socket 15
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.0.125/14515932154768477585
httpd receiving on socket 15
conn_request
Handling request SETUP with URL rtsp://192.168.0.125/14515932154768477585
DACP-ID: 3DFC8E540135ECD1
Active-Remote: 1219664168
Transport: null
type = 110
streamConnectionID = 1134370069952087970
raop_rtp_mirror starting mirroring
Mirroring initialized successfully
httpd receiving on socket 15
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.0.125/14515932154768477585
raop_rtp_mirror accepting client
raop_rtp_mirror width_source = 450.000000 height_source = 972.000000 width = 470.000000 height = 1016.000000
raop_rtp_mirror sps size = 17
raop_rtp_mirror pps size = 4
Got h264 data of 29 bytes
Injecting max_dec_frame_buffering
Video delay is 1584784572616956
raop_rtp_mirror video ntp = 1584784572646023, now = 1584784572617638, latency = -28385
Got h264 data of 5488 bytes
Video delay is -27029
raop_rtp_mirror video ntp = 1584784572662690, now = 1584784572619485, latency = -43205
Got h264 data of 863 bytes
Video delay is -41908
raop_rtp_mirror video ntp = 1584784572712579, now = 1584784572631946, latency = -80633
Got h264 data of 5416 bytes
Video delay is -79331
raop_rtp_mirror video ntp = 1584784572729246, now = 1584784572655644, latency = -73602
Got h264 data of 4104 bytes
Video delay is -71468
raop_rtp_mirror video ntp = 1584784572745913, now = 1584784572669776, latency = -76137
Got h264 data of 7003 bytes
Port settings changed!!
Video pipeline delay is 6 frames or 55903 us
Video delay is -68170
raop_rtp_mirror video ntp = 1584784572762477, now = 1584784572686417, latency = -76060
Got h264 data of 7852 bytes
Video delay is -73083
raop_rtp_mirror video ntp = 1584784572779144, now = 1584784572703539, latency = -75605
Got h264 data of 8587 bytes
Video delay is -74463
raop_rtp_mirror video ntp = 1584784572795810, now = 1584784572719468, latency = -76342
Got h264 data of 10189 bytes
Video delay is -75132
raop_rtp_mirror video ntp = 1584784572812477, now = 1584784572736778, latency = -75699
Got h264 data of 10382 bytes
Video delay is -74536
raop_rtp_mirror video ntp = 1584784572829014, now = 1584784572754118, latency = -74896
Got h264 data of 10158 bytes
Video delay is -73710
raop_rtp_mirror video ntp = 1584784572845681, now = 1584784572769592, latency = -76089
Got h264 data of 9975 bytes
```

pallas, 2020-10-06:
Are you able to run rpiplay under a debugger with `CMAKE_BUILD_TYPE=Debug`?

pallas, 2020-10-06:
FWIW, this is the script I use grab backtraces (renamed `bt`): https://gist.github.com/pallas/035bf6beb9bbf0872d84b38ebacb0cde

pallas, 2020-12-19:
stale

closed 2020-12-19 by pallas