FD-/RPiPlay #77: Autostart RPiPlay

mrwrighty, 2020-03-16:
I have created a script with the following

#!/bin/bash
rpiplay -b auto &

I have then added the location of the script to /etc/rc.local

rpiplay starts as I can see the process in ps -ax, but I am unable to connect to rpiplay from my iPhone.

If I start the script manually everything works as expected.

mrwrighty, 2020-03-16:
OK it appears it is best started as a crontab on reboot rather than from rc.local.

sorriso93, 2020-03-24:
I used systemd and it is ok

iDoka, 2020-03-26:
Looks like need add delay.
Just need sure that all services already started I assume (networking/WiFi/etc).

```
#!/bin/bash
sleep 10
rpiplay -b auto &
```
Works it fine or so ?

alexisteleco, 2020-04-18:
Use systemd. You can ensure the services you need are started, and you are able to wait a certain time too. 

As an example, if you installed RPiPlay in /home/pi, you could create the file `/etc/systemd/system/rpiplay.service` which contains this:
```bash
[Unit]
Description=RPi Play
After=network.target
StartLimitIntervalSec=30

[Service]
Type=simple
Restart=always
RestartSec=10
User=pi
ExecStart=/home/pi/RPiPlay/build/rpiplay -b auto
StandardOutput=inherit
StandardError=journal

[Install]
WantedBy=multi-user.target
```

After that, run this commands and it should work as expected and start at every boot (for me is working OK)
```bash
sudo systemctl daemon-reload
sudo systemctl enable rpiplay
sudo systemctl start rpiplay
```

You can find more information about systemd and its options in [this link](https://www.digitalocean.com/community/tutorials/understanding-systemd-units-and-unit-files)

PaulGriffiths18, 2020-04-19:
Many thanks. I can start RPiPlay ok, have you had any success starting uxplay?   
https://github.com/antimof/UxPlay
I can see the airplay server and connect to it but don’t see any images. On a pi 2 with Stretch

closed 2020-10-06 by pallas

snack862002, 2020-12-29:
> Use systemd. You can ensure the services you need are started, and you are able to wait a certain time too.
> 
> As an example, if you installed RPiPlay in /home/pi, you could create the file `/etc/systemd/system/rpiplay.service` which contains this:
> 
> ```shell
> [Unit]
> Description=RPi Play
> After=network.target
> StartLimitIntervalSec=30
> 
> [Service]
> Type=simple
> Restart=always
> RestartSec=10
> User=pi
> ExecStart=/home/pi/RPiPlay/build/rpiplay -b auto
> StandardOutput=inherit
> StandardError=journal
> 
> [Install]
> WantedBy=multi-user.target
> ```
> 
> After that, run this commands and it should work as expected and start at every boot (for me is working OK)
> 
> ```shell
> sudo systemctl daemon-reload
> sudo systemctl enable rpiplay
> sudo systemctl start rpiplay
> ```
> 
> You can find more information about systemd and its options in [this link](https://www.digitalocean.com/community/tutorials/understanding-systemd-units-and-unit-files)

Maybe coul

> ```shell
> [Unit]
> Description=RPi Play
> After=network.target
> StartLimitIntervalSec=30
> 
> [Service]
> Type=simple
> Restart=always
> RestartSec=10
> User=pi
> ExecStart=/home/pi/RPiPlay/build/rpiplay -b auto
> StandardOutput=inherit
> StandardError=journal
> 
> [Install]
> WantedBy=multi-user.target
> ```

I tried like you described before and something is not working by me. 

pi@raspberrypi:~ $ sudo systemctl daemon-reload
pi@raspberrypi:~ $ sudo systemctl enable rpiplay
Failed to lookup unit file state: Invalid argument
pi@raspberrypi:~ $ sudo systemctl daemon-reload
pi@raspberrypi:~ $ sudo systemctl start rpiplay
Failed to start rpiplay.service: Unit rpiplay.service not found.

Maybe someone explain me why don't work in my case. 

Thanks

pallas, 2020-12-29:
It sounds like you're having a systemd problem, not an rpiplay problem.

snack862002, 2020-12-29:
Thanks for your replay. I used systemd-analyze verify rpiplay.service for more details. Below is the message after command. 

root@raspberrypi:~# systemd-analyze verify rpiplay.service
/lib/systemd/system/nfs-client.target:5: Unknown section 'Service'. Ignoring.
Unit rpiplay.service not found.

Loremwriter, 2022-01-07:
> Use systemd. You can ensure the services you need are started, and you are able to wait a certain time too.
>
> As an example, if you installed RPiPlay in /home/pi, you could create the file [...]

I've tried this one. But when i connect to the RPi via my iPhone, no screen mirror window opens. It just shows the usual ubuntu desktop.