FD-/RPiPlay #82: Only 1ste time RPiPlay works

GoldenEye006, 2020-03-23:
Hello,

Do the corona crysis i've looked a way do Facetime on mine TV, so airplay with mine iPad Pro (2Gen) to the TV would be the easiest way.

So i've took a fresh install of raspbian and build and install RPiPlay on a RPi 3B+
I run the RPiPlay app with no extra command's, (i start the app with a mine computer and with VNC connection to the RPi)

I experience some problems.
- The 1ste time i connect to the app every thing works.
- When i disconnect and later reconnect i've only have video no audio.
- And the 3 time i connect, the RPiPlay app freeze.

When i close RPiPlay and restart, it works but with the same problems like before (1ste time OK, 2 time only video, 3 time freeze)

I would like to run the app 24/7 so i can easy connect without a pc and start the RPiPlay with a computer interface. A RPi is the best solution for 24/7 do to the low power consumption. 

Sorry, mine english isn't that good. I try mine best.

Greetings,
Pieter

sorriso93, 2020-03-24:
Hello same for me. Raspberry PI3, iphone 11 pro and ipad pro and Macbook pro. I can use it the first time, example I connect the iphone then I disconnect it, immediatly after I can connect the ipad and also the mac (disconnecting the ipad). If I try after 1 hour or 2 hour I get only a black screen and an error from any device trying to connect after 1 or 2 minutes.

sorriso93, 2020-03-24:
the service hangs and it does not stop (command executed at 19:11)

pi@RASPBAIRPLAY:/lib/systemd/system $ sudo systemctl restart rpiplay
^C
pi@RASPBAIRPLAY:/lib/systemd/system $ sudo systemctl status rpiplay
● rpiplay.service - Rpiplay airplay server
   Loaded: loaded (/lib/systemd/system/rpiplay.service; enabled; vendor preset: enabled)
   Active: deactivating (stop-sigterm) since Tue 2020-03-24 18:11:17 GMT; 28s ago
 Main PID: 534 (rpiplay)
    Tasks: 11 (limit: 1772)
   Memory: 49.9M
   CGroup: /system.slice/rpiplay.service
           └─534 /rpiplay/build/rpiplay -n TVSALOTTO -b auto

Mar 24 15:18:21 RASPBAIRPLAY systemd[1]: Started Rpiplay airplay server.
Mar 24 15:18:21 RASPBAIRPLAY rpiplay[534]: *** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
Mar 24 15:18:21 RASPBAIRPLAY rpiplay[534]: *** WARNING *** Please fix your application to use the native API of Avahi!
Mar 24 15:18:21 RASPBAIRPLAY rpiplay[534]: *** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Mar 24 15:18:21 RASPBAIRPLAY rpiplay[534]: *** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
Mar 24 15:18:21 RASPBAIRPLAY rpiplay[534]: *** WARNING *** Please fix your application to use the native API of Avahi!
Mar 24 15:18:21 RASPBAIRPLAY rpiplay[534]: *** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Mar 24 18:11:17 RASPBAIRPLAY systemd[1]: Stopping Rpiplay airplay server...

sorriso93, 2020-03-24:
Additional info. After restart I have memory reserved for the service about 1.2M, I connect the iphone and then disconnect it I get 5.8M, same with ipad 8.8M, after connecting the mac 15.3M. Before the restart I had 49.9M of allocated memory and it continue increasing with no connection. **It seems there is something that continue allocating memory without freeing it**

pallas, 2020-04-30:
I've noticed this too. Currently, I run a script that checks to see if there is an active connection, and then once that connection dies it kills rpiplay. It also starts/stops my screensaver. See below.

```
#!/bin/bash
# Author: Derrick Pallas <derrick@pallas.us>

xset s off
xset -dpms
xset s noblank

modprobe snd_bcm2835 && aplay /usr/share/sounds/alsa/Noise.wav

nice -n 5 /usr/lib/xscreensaver/feedback --root &

while true ; do
        killall -SIGTERM rpiplay || true
        killall -SIGCONT feedback || true
        killall -SIGHUP rpiplay || true ; usleep 0.1
        killall -SIGKILL rpiplay || true

        env DISPLAY=:0.0 nice -n -15 rpiplay -n Espejo -b auto -l 2>&1 &

        while ! lsof -i -sTCP:ESTABLISHED | grep rpiplay >/dev/null ; do
                sleep 1
                pidof rpiplay || continue 2
        done

        # echo on 0 | cec-client RPI -s -d 1
        # echo as | cec-client RPI -s -d 1
        killall -SIGSTOP feedback || true

        while lsof -i -sTCP:ESTABLISHED | grep rpiplay >/dev/null ; do sleep 1 ; done

        # echo is | cec-client RPI -s -d 1
done

#
```

This script is the argument given to `xinit` on my box. Note the lines that do CEC. I went back and forth on whether I liked it or not but the TV had this weird modal when coming out of CEC and my kid kept stealing the TV from the other room. YMMV.

privaloops, 2020-05-02:
Same here : Raspberry PI4, iphone 11, after stopping screen share, the screen is black with the cursor available. If I quit RPiPlay desktop come back.

pallas, 2020-10-06:
I would very much suggest adding a script like the above that lets rpiplay run for one client and then restarts. Going to close due to staleness, & a proposed more-or-less working xinit script.

closed 2020-10-06 by pallas