FD-/RPiPlay #84: Audio problem when stealing display, maybe other times

pallas, 2020-03-25:
If I have my Macbook Air connected, multiple audio streams usually work (eg: Youtube in Firefox, Apple Music, & Videos from Photos; not always though). If I then steal the AirPlay service using my iPad and steal it back with my MBA, video works but not audio.

This report is for reference, I'm currently doing some debugging here. Any tips are welcome.

pallas, 2020-03-25:
It looks like the difference is that the latency numbers are crazy when it doesn't work and sane when it does.

FD-, 2020-03-25:
Yeah, I had seen that as well a while back, found that really puzzling, but couldn't find the cause.

pallas, 2020-03-25:
Whatever it is, low latency mode helps. I think the latency offset is getting pushed far into the future.

~Derrick • iPhone

pallas, 2020-05-09:
This is happening with video too. For instance, I rotated my iPhone, which started a new h264 stream but kept the old latency calculation. So the video latency went from -70k to +1.7M, hanging the display.

```
Got h264 data of 1019 bytes
Video delay is -47036
raop_rtp_mirror video ntp = 1589040900035758, now = 1589040899988977, latency = -46781
Got h264 data of 1029 bytes
Video delay is -45928
raop_rtp_mirror video ntp = 1589040900152479, now = 1589040900073681, latency = -78798
Got h264 data of 1020 bytes
Video delay is -77832
raop_rtp_mirror video ntp = 1589040900169146, now = 1589040900327107, latency = 157961
Got h264 data of 1037 bytes
httpd receiving on socket 15Video delay is 1030715

conn_request
raop_rtp_mirror video ntp = 1589040900185813, now = 1589040901468486, latency = 1282673
Got h264 data of 1016 bytes
Handling request POST with URL /feedback
raop_handler_feedback
Video delay is 1737763
raop_rtp_mirror video ntp = 1589040900202479, now = 1589040903047980, latency = 2845501
httpd receiving on socket 15
Got h264 data of 1030 bytes
Video delay is 13581328
raop_rtp_mirror video ntp = 1589040900519144, now = 1589040924286864, latency = 23767720
conn_requestGot h264 data of 2485 bytes

Video delay is 43618256
raop_rtp_mirror video ntp = 1589040900535820, now = 1589040955221983, latency = 54686163
Got h264 data of 1332 bytes
Video delay is 73046018
Handling request POST with URL /feedback
raop_rtp_mirror video ntp = 1589040900552520, now = 1589040984925429, latency = 84372909
raop_handler_feedback
```

pallas, 2020-05-09:
Right now I added some code to reset `first_packet_time` for both the video and audio renderers when the delay is greater than some small threshold. Note, that this happens when you put a mirroring phone to sleep and then wake it up, so it's pretty easy to test.

FD-, 2020-05-09:
Does it fix the audio problem?

pallas, 2020-05-09:
Yeah, it does. Patch incoming.

pallas, 2020-05-09:
Note, it also seems to resolve a video hang I'm able to reproduce.

pallas, 2020-10-06:
Resolved.

closed 2020-10-06 by pallas