FD-/RPiPlay #86: AirPlay mirroring won’t disconnect

naorth, 2020-03-25:
Hi,
When I connect to the airplay server from my phone on iOS 13.4, everything works fine. However, when I try to disconnect, on my phone it will appear as disconnected but it’s still mirroring to the pi. After a minute or so the video stream crashes and I have to reboot the pi as I can no longer airplay to it.
Any ideas how to resolve this issue?

pallas, 2020-04-30:
I've seen this as well.

Mrc527, 2020-05-05:
Yep, that's happening on macOS Catalina (10.15.4) as well. Once you click "disconnect" it takes almost 1 minute to actually disconnect. The server looks frozen with the last image from my mac and it won't accept any new connection till I forcefully restart it (kill -9)

tenbergen, 2020-05-24:
Same here. Running on RPi 3b with buster light.

pallas, 2020-05-24:
As of my last patchset, I've seen this improve. My guess is that adjusting the h264 stalls & not blocking on buffers (except for the final flush) helped. I've also disabled the h264bitstream hack to lower the pipeline latency because I don't believe the h264bitstream library is safe. YMMV.

If you'd like to try it, I can push my current build to my fork.

pallas, 2020-05-24:
@tenbergen @Mrc527 @naorth please see https://github.com/pallas/RPiPlay/pull/new/running-branch

tenbergen, 2020-05-24:
Hi! Thanks for the quick reply. Just tried your branch on fresh OSMC 2020.3 install on RPi 3b. Sadly, it won't build. Cmake success, followed by make:

[ 23%] Building C object lib/CMakeFiles/airplay.dir/raop.c.o
In file included from /home/osmc/RPiPlay/lib/raop.c:67:0:
/home/osmc/RPiPlay/lib/raop_handlers.h:22:25: fatal error: plist/plist.h: No such file or directory
 #include <plist/plist.h>
                         ^
compilation terminated.
lib/CMakeFiles/airplay.dir/build.make:350: recipe for target 'lib/CMakeFiles/airplay.dir/raop.c.o' failed
make[2]: *** [lib/CMakeFiles/airplay.dir/raop.c.o] Error 1
CMakeFiles/Makefile2:296: recipe for target 'lib/CMakeFiles/airplay.dir/all' failed
make[1]: *** [lib/CMakeFiles/airplay.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Sidenote: on OSMC, also needed to install gcc cmake make build-essential first, then copy /opt/vc/include and /opt/vc/src from https://github.com/raspberrypi/firmware/releases (I used the Feb 12 release).

FD-, 2020-05-24:
@pallas' branch relies on a separate libplist install. I think running 'sudo apt-get install libplist' should do it.

pallas, 2020-05-24:
That's correct, I forgot to add that to the dependency list.

tenbergen, 2020-05-24:
Thanks! libplist-dev worked and I was able to make.

Ran it with OSMC concurrently. Same issue as before. Moreover, I get:
aacDecoder_DecodeFrame error : 0x4002
aacDecoder_DecodeFrame error : 0x400c

Trying to disconnect gives:

aacDecoder_DecodeFrame error : 0x400c
Connection closed for socket 15
Destroying connection

Can't reconnect after. My iPhone just sits there thinking. After a minute, it says the "connection to *server name* could not be established."

cntrl+c won't really stop the program. It says "Stopping..." which won't go away. Have to kill -9 processID.

tenbergen, 2020-05-24:
Oh, also: no sound. It steals the display from OSMC, cell phone says it's playing, but no audio comes out of HCMI nor analog.

FD-, 2020-05-24:
Are you sure you connect to RPiPlay via the AirPlay Mirroring control in Control Center and not the AirPlay Speaker control? These are different protocols and RPiPlay only supports mirroring.

tenbergen, 2020-05-24:
Oh! I didn't realize that. You're right, using "mirroring," it works now. Multiple connections work fine, stopping the server with ctrl+c works also. Image goes back to OSMC after stopping the server.

Picture quality is outstanding, video streams very quickly. Well done!

The sound is garbled, however. It sounds like the codec issues I get from certain (perfectly legally) downloaded movies. But this might be a different issue.

Conclusion: @pallas 's patch solves connection issues on my machine.

pallas, 2020-10-06:
Resolved.

closed 2020-10-06 by pallas