FD-/RPiPlay #88: No audio!

obaidsamadian, 2020-03-25:
I have two routers (Huawei and MikroTik).
when I connect the Pi to the Huawei Router and  I mirror the screen audio works fine, but when I connect the Pi to the MikroTik router the audio doesn’t work.

Lineaire, 2020-03-28:
I have the same problem as you, on my mikrotik it does not work. probably a wearing problem.

obaidsamadian, 2020-03-29:
I've installed and configured multicast package on my mikrotik. I didn't work.

> I have the same problem as you, on my mikrotik it does not work. probably a wearing problem.

obaidsamadian, 2020-04-01:
> I have the same problem as you, on my mikrotik it does not work. probably a wearing problem.

I found a solution! I've made these changes on my Pi: 
1- Increase GPU memory
2- Change display resolution to 720p 50hz
3- add latency flag ( -l ) to rpiplay

g-philipp, 2020-05-01:
Ive got the same problem and Im getting the following error: aacDecoder_DecodeFrame error : 0x400c

How much did you increase the GPU memory.

Btw Im using an usb dac and I changed the audio device in this config: "nano /usr/share/alsa/alsa.conf" (Yes normally the audio is working fine, but not with AirPlay or screenmirroring)

FD-, 2020-05-02:
The decode error usually means you're using RPiPlay as an audio-only Airplay receiver, which it doesn't support. It only supports screen mirroring. See #2.

obaidsamadian, 2020-05-02:
> Ive got the same problem and Im getting the following error: aacDecoder_DecodeFrame error : 0x400c
>
> How much did you increase the GPU memory.
>
> Btw Im using an usb dac and I changed the audio device [...]

400

FD-, 2020-05-02:
@g-philipp Also, I missed you're using a USB DAC. These are not supported by RPiPlay, because the hardware PCM decoder only supports HDMI and analog audio.

closed 2020-05-25 by FD-

brechtknecht, 2021-12-13:
I figured out how to get audio working with my setup. I noticed that when I was running the `rpiplay` server with `-ar gstreamer` an error occured.

The fix was to install a missing package, answer from [here](https://forums.raspberrypi.com/viewtopic.php?t=140706):

`sudo apt-get install gstreamer1.0-alsa`

Now my audio is working, both hdmi and analog.