FD-/RPiPlay #89: Documentation clarification

jamieburchell, 2020-03-25:
In the README it states:

> Compile with -O3 (`cmake --DCMAKE_CXX_FLAGS="-O3" --DCMAKE_C_FLAGS="-O3" ..`)

What is the performance benefit from doing this and what is the full command (what is ...) ? Is this purely to speed up the compiling, or does it affect the performance of the application?

Can the apt installed components and required /opt/vc files be removed once the binary is compiled?

Thanks
Jamie

FD-, 2020-03-26:
-O3 is a compiler flag that produces better performing binaries at the expense of longer compile time. The complete command is what's between the parentheses. On a Pi Zero, for me it made the difference between somewhat useable audio performance and usable audio performance.

I guess you could uninstall cmake after the build, but all library dependencies (e.g libavahi and libssl) are required for running RPiPlay.

closed 2020-03-26 by FD-

jamieburchell, 2020-03-26:
Thanks for the clarification. I only just realised that ".." was parent directory rather than ellipsis.