FD-/RPiPlay #90: Blurry image and small images centered

Tauri28, 2020-03-29:
I have set up RPiPlay server on my Raspberry Pi 3+. Mirroring with iphone 11 pro. Mirroring to Samsung 55" 4k TV. Everything is working but having 2 issues:

1. The mirrored content on the TV is blurry.
2. If I view images, they cover only approximately 25% of the tv screen, are in the center of the screen surrounded by black borders - they are sharp tho. 

Is it caused by the upscaling from iphone resolution to tv resolution?

pgeler, 2020-08-28:
Similar issue with PI3 and MacOSX 🤔  : 1080p

pallas, 2020-08-28:
On my MBA, I had set it to send images defaulting to the display size. I have an older Samsung and I had to disable overscan on it as well. Let me see if I can figure out the incantation for that.

pallas, 2020-08-28:
Ok, on the MacOS side, once you're connected go to Display Preferences and set it to either optimize for the built-in display or for the airplay display, and make sure it's not set to scaled. On the Samsung, YMMV but if you click SOURCE > TOOLS, you can edit the device type and set it to PC or DVI PC. I thought there was a menu item under Menu > Picture for me, but I don't see it now.

pgeler, 2020-08-28:
my TV is also pretty old, so I was already using DVI mode, when trying to switch between modes (retina vs rpiplay) it looks like almost the same, maybe in retina optimize it is even better, (so nothing change in blur, it almost seem not as a blur, but something like when screen is running in not "native resolution")
Same TV with HDMI cable or "airplay server" thought windows seems fine. Maybe need to make a photo of selected area in tree different approaches... native resolution for TV is 1080p 1920x1080@60, also will try to run it in ECA mode

pallas, 2020-10-06:
Were you able to test your hypothesis?

seungjunProgramming, 2020-11-27:
raspberrypi 4B + Macos 11.0.1 Same too

pallas, 2020-12-19:
this seems like a general monitor or pi configuration issue, not specific to rpiplay

closed 2020-12-19 by pallas