FD-/RPiPlay #91: Firewall ports to allow screen mirroring?

roflmaooooo, 2020-04-01:
Works fine with my firewall turned off but I run a Pi4 with a restrictive firewall so I would like to know what ports should I open to enable screen mirroring. Might be helpful to many if you can include them in the documentation as well. Thanks for the great work btw!

pimmler123, 2020-04-13:
Hi, Here are some Ports used by AirPlay:
https://nto.github.io/AirPlay.html#video
Maybe it worked if you open them.

inferiorhumanorgans, 2020-05-21:
The way it's currently designed you'll need to open everything because RPiPlay lets the operating system pick random local ports.  If you were to hardcode the local ports you'd still need to open up everything outbound to the iDevice for audio synchronization to work.

pallas, 2020-10-06:
Out of scope?

closed 2020-10-06 by pallas