FD-/RPiPlay #94: About not being able to stop

vinicyusmacedo, 2020-04-06:
Hey there, thanks for the tool! I wish you did it before I bought an Apple TV (but due to the quarantine, I don't have it with me, so it's coming out handy)

Whenever I try to Airplay an Youtube video (which won't work , see #7) RPiPlay hangs and I can't exit, I have to SIGKILL it. I've made some debugging and found out that [this call](https://github.com/FD-/RPiPlay/blob/01279bb2ae582d126f831c21ba019c4865f3391d/renderers/video_renderer_rpi.c#L454) is the one that hangs when you want to exit rpiplay. 

Looking at [here](https://github.com/adafruit/pi_hello_video/blob/e08d2a43a9cbaa38ba0decb86f972b558ee7eef4/libs/ilclient/ilclient.h#L756), it appears that the last parameter blocks the function until a buffer is available, which might not be the case and it will wait for it infinitely.

pallas, 2020-05-08:
I'm seeing a stack corruption with a duplicate return value of `AES_set_encrypt_key`, so possibly near a call to that function. Unfortunately when this happens, the system hangs so I can't `gdb` it live.

pallas, 2020-10-06:
Can you please try with the latest upstream version, or with my future version at https://github.com/pallas/RPiPlay/tree/running-branch ?

pallas, 2020-12-19:
stale

closed 2020-12-19 by pallas